pikvm/pikvm #1159: Make mouse poll frequency default, minimum, and maximum a configurable override

dyehya, 2023-11-17:
**Is your feature request related to a problem? Please describe.**
There are times when I would prefer extremely low mouse polling frequency and other times where a much higher frequency may be required. It is also very important that I can tweak the upper and lower bounds for users in specific environments.

**Describe the solution you'd like**
Be able to use override.yaml to configure the number of ms for the mouse polling default value, minimum, and maximum available in the UI.

```
kvmd:
    hid:
        mouse_poll:
            default: 10
            max: 5000
            min: 1
```

**Describe alternatives you've considered**
None

**Additional context**
While 10ms is a much better experience, there are times when fast, high precision movement/perceived movement is more important or feasible depending on a variety of factors including network considerations so being able to tweak the default and lower or increase this even further or limit the ability of the user to do so is very desirable.

mdevaev, 2023-11-17:
The browser remembers your value, isn't that enough?

dyehya, 2023-11-17:
No because I will (eventually) have dozens of different users (depending on how successful these pilots are) using different remote machines each time. The ability to change default value is the more critical change is more critical as that is something I would be able to control and not rely on the users to remember as they use a different browser.

With the limited use so far, when the default was 100ms I had to remind the people I gave access to a few times each before they got into the habit of reducing it to 10ms on different browsers/devices. Also, there may be some cases where I want it down to 1ms vs the 10ms minimum but I don't think it is a great idea to change the minimum value to 1ms for everyone but as an override that would be acceptable. The max value I included because of the other 2 but isn't a big deal for my use cases.

10ms is a good balance overall but there is a learning curve to users with the platform and the options available so the more I can default, on specific PiKVMs the better. Another case I had is I needed to share/record the screen and wanted it to be "buttery smooth" so even 10ms is not the greatest so 1ms minimum temporarily would have been great. At 10ms it wasn't super noticeable but when quickly moving the mouse across the screen there were a couple times when it "jumped".

Also, because I am not demanding enough :sarcasm:, it would be nice if the default value option is added as an override, that the KVMD default that is cached in the browser be cleared/reset on default change. I updated the original post desired behavior section with that.

Thanks.