pikvm/pikvm #1170: janus.js on Google Chrome - black screen, or streaming cuts off periodically

Majestic7979, 2023-11-23:
I am reporting from Windows 11. On Google Chrome I keep getting black screens or the stream cuts out periodically. I noticed that janus.js (on developer tools) keeps saying that ICE disconnected.

In Firefox I get normal connection. The issue is that some of my clients block anything but Google Chrome on their devices so I cannot use anything but Google Chrome and this is making the PiKVM unusable in those situations causing massive headache, clients are unhappy.

I don't know if this is something related to Chrome or if there might be a quirk in PiKVM that is not playing along with Chrome.

Any ideas or solutions please? I use Firefox when possible but it's not always possible.

[image]

Majestic7979, 2023-11-27:
I noticed that at the top of the window where it shows h.264 + audio, when I get a cutout it will say "Invalid Peer Connection" briefly.

plia7, 2023-11-28:
@Majestic7979 It sounds like it could be related to the issue that I reported here:
https://github.com/pikvm/pikvm/issues/1084
@mdevaev

mdevaev, 2023-12-03:
Please attach the full output of `journalctl -u kvmd-janus`. You can remove your IPs from it if you want.

mdevaev, 2023-12-04:
The problem of @plia7 xxx is similar to the other, and manifests itself in other, more difficult conditions.

Did I understand you correctly that if you use H264 WITHOUT audio, then the problem does not play?

Majestic7979, 2023-12-05:
Yes that is correct, without audio (volume slider on 0%) it does not seem to happen. I will test more in the next couple of hours and confirm for sure.

plia7, 2023-12-06:
> The problem of @plia7 xxx is similar to the other, and manifests itself in other, more difficult conditions.
@mdevaev My problem is only happening from an external network. On the local network it's working fine.
It's strange that @Majestic7979 problem is supposedly happening on the local network, maybe he thinks it's happening on the local network but it's actually happening on the external network like mine which it goes through when the issue happens? @Majestic7979 Try maybe to remove Tailscale or disable it to ensure it doesn't go through the external network, to ensure it's actually happening on the local network.

Majestic7979, 2023-12-09:
I am opening the PiKVM in my browser with its internal 192.168.x.x IP... it only happens in Google Chrome. Firefox is fine. I don't have Tailscale, there's nowhere in any of my comments talking about it. Why are you making this assumption!? Clearly wrong one because you assumed I have Tailscale when I never said anything about it...

Instead_ of assuming things please ask me first as we do not want to muddy my issue with wrong information, thank you. Assumptions are terrible, @mdevaev is a busy person, we don't want to waste his precious time with "what ifs" and it also wastes your time because you are thinking of a solution for my issue that does not even have one of the factors...

The issue happpens all the time but only on Google Chrome. The only thing I could think of is that in my kvmd config file I have enabled USB Networking, this is because sometimes I am unable to be present, so I have an arrangement with the client for one of their trusted employees to go in their server room and plug their laptop into the PiKVM, which will then pass through ethernet through USB to their laptop and then I can download support files etc more easily than setting up firewall rules through their network. I don't know if this would affect anything. Even when I am connected to their laptop it happens.

```
otgnet:
    firewall:
        allow_tcp: [80, 443]
        forward_iface: eth0
    commands:
        post_start_cmd_append:
            - "--dhcp-option=6,8.8.8.8,8.8.4.4"
    iface:
        ip_cmd:
            - /usr/bin/ip
        net: 10.65.0.0/28
```
That is the only thing that I think of that could be causing problems!?

But then why only on Google Chrome?

mdevaev, 2023-12-09:
I'm currently trying to find a way to reproduce at least one of your problems. It seems to me that there is one root here, but I'm not sure.

Majestic7979, 2023-12-10:
> I'm currently trying to find a way to reproduce at least one of your problems. It seems to me that there is one root here, but I'm not sure.

Thank you for trying, let me know how I can help within my possibilities

mdevaev, 2024-03-11:
Okay, this is the same problem as #1084, so I'm closing it.

closed 2024-03-11 by mdevaev