pikvm/pikvm #1175: Temperatures on GUI

mainmind83, 2023-11-30:
**Is your feature request related to a problem? Please describe.**
From the console it is possible to access the data, it would be interesting to be able to see it from the graphical interface

**Describe the solution you'd like**
Show current CPU and GPU temperatures, Little historical graph?
Show temperature alert in the same way as the low voltage alert.
Flash with temperature warning, if display I2C is connected

**Describe alternatives you've considered**
vcgencmd measure_temp
echo $(</sys/class/thermal/thermal_zone0/temp)/1000 | bc

mainmind83, 2023-12-01:
And add GPU temperature or little historical graph?

mdevaev, 2023-12-01:
The GPU is on the same chip as the CPU and the difference is negligible.

As for graph, this function does not relate directly to KVM, and given the automatic processing of temperature increases, I don't think this is necessary. Moreover, we don't have the ability to save this information anywhere between reboots. Any graph drawing feature that I add will be worse than the external monitoring that you can use.

* https://docs.pikvm.org/prometheus
* https://<pikvm>/api/info

closed 2023-12-01 by mainmind83