pikvm/pikvm #1177: Audio Slider does not show up after following instructions

d7laungani, 2023-11-30:
**Describe the bug**
Audio Slider does not show up on UI after following instructions in 
https://docs.pikvm.org/audio/

**To Reproduce**
"/etc/kvmd/janus/janus.plugin.ustreamer.jcfg"

video: {
    sink = "kvmd::ustreamer::h264"
}
audio: {
    device = "hw:0,0"
    tc358743 = "/dev/kvmd-video"
}

 cat /proc/asound/cards

 0 [tc358743       ]: simple-card - tc358743
                      tc358743

/boot/config.txt

# See /boot/overlays/README for all available options
initramfs initramfs-linux.img followkernel

hdmi_force_hotplug=1
gpu_mem=128
enable_uart=1
dtoverlay=tc358743
dtoverlay=disable-bt
dtoverlay=dwc2,dr_mode=peripheral
dtparam=act_led_gpio=13

# HDMI audio capture
dtoverlay=tc358743-audio

# SPI (AUM)
dtoverlay=spi0-1cs

# I2C (display)
dtparam=i2c_arm=on

# Clock
dtoverlay=i2c-rtc,pcf8563

**Expected behavior**
Audio Slider should show up on UI but it is missing and so I cannot hear and HDMI audoio

**Screenshots**

**Desktop (please complete the following information):**
 - OS: MACOSX
 - Browser Chrome
 - Version 118.0.5993.96
browser

**PiKVM info:**
  - Raspberry Pi board version RPI 4
  - PiKVM platform v3
  - Video capture type 
  - KVMD version: 3.265-1
  - uStreamer version: `5.44-2`
  - Linux kernel: `6.1.57-2-rpi-ARCH `

**Additional context**
Add any other context about the problem here.

mdevaev, 2023-12-01:
Hello. What do want to do?

Also attach the log `journalctl -u kvmd-janus`

mdevaev, 2023-12-03:
Hello?