pikvm/pikvm #1182: disabling https redirect cause 404 not found error.

dowonim323, 2023-12-05:
worker_processes 4;

# error_log /tmp/kvmd-nginx.error.log;
error_log stderr;

include /usr/share/kvmd/extras/*/nginx.ctx-main.conf;

events {
    worker_connections 1024;
    use epoll;
    multi_accept on;
}

http {
    types_hash_max_size 4096;
    server_names_hash_bucket_size 128;

    access_log off;

    include /etc/kvmd/nginx/mime-types.conf;
    default_type application/octet-stream;
    charset utf-8;

    sendfile on;
    tcp_nodelay on;
    tcp_nopush on;
    keepalive_timeout 10;
    client_max_body_size 4k;

    client_body_temp_path   /tmp/kvmd-nginx/client_body_temp;
    fastcgi_temp_path       /tmp/kvmd-nginx/fastcgi_temp;
    proxy_temp_path         /tmp/kvmd-nginx/proxy_temp;
    scgi_temp_path          /tmp/kvmd-nginx/scgi_temp;
    uwsgi_temp_path         /tmp/kvmd-nginx/uwsgi_temp;

    include /etc/kvmd/nginx/kvmd.ctx-http.conf;
    include /usr/share/kvmd/extras/*/nginx.ctx-http.conf;

    server {
        include /etc/kvmd/nginx/listen-http.conf;
        include /etc/kvmd/nginx/certbot.ctx-server.conf;
        # include /etc/kvmd/nginx/redirect-to-https.conf;
    }

    server {
        include /etc/kvmd/nginx/listen-https.conf;
        include /etc/kvmd/nginx/ssl.conf;
        include /etc/kvmd/nginx/kvmd.ctx-server.conf;
        include /usr/share/kvmd/extras/*/nginx.ctx-server.conf;
    }
}

Like above, I commented "include /etc/kvmd/nginx/redirect-to-https.conf;" this part, but i get 404 not found error.

I have read pikvm handbook about same problem, but i could not get the point.

Disable HTTPS at all in /etc/kvmd/nginx/nginx.conf. To do this, comment some lines [like in this file](https://github.com/pikvm/kvmd/blob/master/configs/nginx/nginx.conf#L39) and restart web server: systemctl restart kvmd-nginx.

This is what handbook says, but there was no clue about lines to comment on link "this file"

I think the conf file is updated after the handbook is written.

mdevaev, 2023-12-05:
Hello. You need to keep only one server section with following content:
```nginx
    server {
        include /etc/kvmd/nginx/listen-http.conf;
        include /etc/kvmd/nginx/certbot.ctx-server.conf;
        include /etc/kvmd/nginx/kvmd.ctx-server.conf;
        include /usr/share/kvmd/extras/*/nginx.ctx-server.conf;
    }
```

dowonim323, 2023-12-05:
> Hello. You need to keep only one server section with following content:
>
> ```nginx
>     server {
>         include /etc/kvmd/nginx/listen-http.conf;
>         include [...]

I fixed it. Thank you!

closed 2023-12-05 by dowonim323

mdevaev, 2023-12-05:
:ok_hand: