pikvm/pikvm #1183: Any recommendation for hdmi csi-2?

haze1986, 2023-12-05:
Dear all, I'm tempted to try with newer revisions of the hdmi csi-2 that may allow for 1080p 60hz. Does anyone know other than the offical, is there a list of working modules?

mdevaev, 2023-12-05:
There are no modules for RPi2-4 supporting 60Hz at all, because 1080p50Hz is a physical limitation of the bus, that is, the number of wires inside the connector. In order to implement 60Hz, we had to develop a V4 with our own circuitry. So if you really need 60Hz, then you need V4.

mdevaev, 2023-12-05:
I do not recommend using bli* at all, because these are, in fact, low-quality basement fakes. According to reports in the discord, they burn out, simply do not boot, suffer from poor design (the official recommendation is to seal part of the board with duct tape to avoid a short circuit with the CM4 module), and their chips fall off due to poor soldering, etc. Well, not to mention that actually bli* steal our schematics, send spam, use my software and give nothing in return, trying to expose themselves as the original authors of open source kvm. Do you really want to support these guys? :)

The V4 has its price not because we are greedy, but to ensure the reliability of the device itself. We use automation grade components, test batches at the factory, test boards, do X-ray inspection, test the finished device, and so on. You are buying a device for emergency remote access, it must be reliable, once and for a long time. If a miracle happens and our device turns out to be defective, you will have no problems changing it. And the broken device will go to our laboratory and be dissected to find out the cause and add another factory test to prevent this in the future.

haze1986, 2023-12-05:
> I do not recommend using bli* at all, because these are, in fact, low-quality basement fakes. According to reports in the discord, they burn out, simply do not boot, suffer from poor design (the official recommendation is to seal part of the board with duct tape to avoid a short circuit with the CM4 module), and their chips fall off due to poor soldering, etc. Well, not to mention that actually bli* steal our schematics, send spam to discord pm with advertising, use my software and give no contributions in return, trying to expose themselves as the original authors of open source kvm. Do you really want to support these guys? :)
> 
> The V4 has its price not because we are greedy, but to ensure the reliability of the device itself. We use automation grade components, test batches at the factory, test boards, do X-ray inspection, test the finished device, and so on. You are buying a device for emergency remote access, it must be reliable, once and for a long time. If a miracle happens and our device turns out to be defective, you will have no problems changing it. And the broken device will go to our laboratory and be dissected to find out the cause and add another factory test to prevent this in the future.

Understood, I happen to want to test this just cos I have a spare pi zero 2 and would like to absolute minimum cost. Thank you for your clarifications though.

mdevaev, 2023-12-05:
As I said, there is a physical limitation and cannot be done without a whole small motherboard around compute module. A couple of days ago, [I updated the instructions for DIY solutions](https://docs.pikvm.org/v2). If you want the cheapest possible testing, then just buy any compatible module. Yes, you'll be limited to 1080p 50Hz, but it still works great.

haze1986, 2023-12-05:
> As I said, there is a physical limitation and cannot be done without a whole small motherboard around compute module. A couple of days ago, [I updated the instructions for DIY [...]

Is there a list of compatible modules? The link for DIY solutions shows the waveshare model with 1080p30.

Thanks.

mdevaev, 2023-12-05:
All modules are compatible except Auvidea B101, but you are unlikely to buy it somewhere. Zero only requires a special CSI cable, I described this in the instructions.

closed 2023-12-05 by mdevaev