pikvm/pikvm #1184: Use Arduino Micro as HID Keyboard/Mouse?

deanfourie1, 2023-12-06:
**Is your feature request related to a problem? Please describe.**
No, more of a question?

**Describe the solution you'd like**
Is it possible to use a Arduino Micro for Keyboard/Mouse Emulation?

**Additional context**
Any other recommendations to make the keyboard mouse emulation work cleanly?

nblky, 2023-12-06:
https://github.com/JDecelles1990/PiKVM/blob/main/pages/arduino_hid.md

mdevaev, 2023-12-06:
There is a [technical possibility](https://docs.pikvm.org/arduino_hid/) to do this, but it is deprecated and not recommended for use.

Due to the voltage difference between Arduino and Raspberry, a logic level shifter is required, and in general the schematic is more complex than with Pico. [All official instructions and further updates will be for Pico only](https://docs.pikvm.org/pico_hid/). Pico also leaves the UART free for console.

[image]

closed 2023-12-06 by mdevaev