pikvm/pikvm #1188: Unable to see UEFI BIOS on ASRock Z790M-ITX

CloudMass, 2023-12-11:
**Describe the bug**
I have build a NAS using the ASRock Z790M-ITX motherboard but I can't see the BIOS when using the PiKVM

**To Reproduce**
Steps to reproduce the behavior, like:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Expect to see a BIOS boot image with option to enter the BIOS and make changes

**Screenshots**
N/A

**Desktop**
 - OS: Ubuntu 22.04.3 LTS
 - Browser: Brave 1.59.117

**PiKVM info:**
  - Raspberry Pi board version: Raspberry Pi 4 Model B Rev 1.1
  - PiKVM platform: Ready-made PiKVM V3
  - KVMD version: kvmd 3.276-1, kvmd-fan 0.26-1, kvmd-oled 0.26-1, kvmd-platform-v3-hdmi-rpi4 3.276-1, kvmd-webterm 0.44-1
  - uStreamer version: ustreamer 5.45-1
  - Linux kernel: Linux pikvm 6.1.61-1-rpi-ARCH #1 SMP Fri Nov  3 20:48:52 MSK 2023 armv7l GNU/Linux

**Additional context**
I have tried several different EDID settings both with HDMI -> HDMI & DP -> HDMI.
The [default suggested](https://docs.pikvm.org/edid/#default-edid as well as some custom 1024x768, 1280x720 & 1920x1080)

It will never show the BIOS but will show the video output just fine as soon as the OS boots.

arch1mede, 2023-12-11:
Hi, we would recommend that you join [Discord](https://discord.gg/bpmXfz5) for further troubleshooting, you may need to try all of the EDID's as we don't have any specific info on which EDID is going to work with your MB BIOS.

IF no EDID works, these its possible that you will need to also get some down scaler to get BIOS to show up, we would recommend the [JTECH](https://www.amazon.com/Splitter-Multi-Resolution-J-Tech-Digital-downscale/dp/B0839PRQ2V?th=1).

mdevaev, 2023-12-14:
I'm closing this because we're dealing with these kinds of issues in discord. If you haven't written there yet, please do so, and our support will help you.

closed 2023-12-14 by mdevaev

CloudMass, 2023-12-27:
To anyone having this issue I ended up solving it by using a HDMI EDID Emulator.