pikvm/pikvm #1190: GPIO not available

Venkateshgithubb, 2023-12-14:
When I clone kvmd from the github in ubuntu and try to build it using the sudo make command. I am facing a major issue.
Issue-"sudo modprobe gpio-mockup gpio_mockup_ranges=0,40
modprobe: FATAL: Module gpio-mockup not found in directory /lib/modules/6.2.0-37-generic
make: *** [Makefile:98: /dev/gpiochip0] Error 1"
I am facing the above issue while building the kvmd repo in ubuntu linux machine.

closed 2023-12-20 by mdevaev

stappersg, 2025-01-31:
For those who come here also through a _websearch_ like me: https://github.com/WangHanChi/gpio-mockup

stappersg, 2025-02-01:
And later I found https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/gpio/gpio-mockup.c

(Yes, I had reasons for looking further after my previous find. ( https://github.com/WangHanChi/gpio-mockup/issues/7 ) )