pikvm/pikvm #1191: Windows HID not working?

deanfourie1, 2023-12-15:
**Describe the bug**
The PICO HID device is not recognized by windows. Instead, it shows in device manager as a RP2 Boot device and the keyboard and mouse does not work on target machine. I tested this previously on a different machine and it worked without issue.

The only real difference is the previous machine had internet, this one does not. But I have tried it with internet too to see if it will download a driver, it does not.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Using a PICO, setup as per the instructions, plug in USB and no mouse/keyboard control.

**Expected behavior**
keyboard and mouse to work.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser: Chrome/Edge

**PiKVM info:**
  - Raspberry Pi board version: rpi4
  - PiKVM platform: v2
  - Video capture type: HDMI USB Capture
  - KVMD version: `pacman -Q | grep kvmd`

**Additional context**
Add any other context about the problem here.

mdevaev, 2023-12-15:
Are you sure that you did not press the reset button on the board when connecting? If this happens, then Pico will remain in boot mode. Try to completely disconnect the KVM from the host (including power) and reconnect it.

deanfourie1, 2023-12-15:
I'm sure, the the unit is completely boxed.

Thanks

[image]

[image]

mdevaev, 2023-12-16:
You have two devices with Pico, as I see it. Does the second one behave exactly the same, or is the problem only observed on one?

deanfourie1, 2023-12-16:
Yes, both behave the same.

This is strange, as I tested both these devices prior to deployment and they worked. 
I then took the device out that is not working and tested it again on the same machine I first tested on, and same result. It just showed up in device manager as RP2 boot and did not work.
I dropped the firmware back on the pico as it was showing up as a USB drive, so i take it it was in flash mode? Anyway, I dropped it back on, it rebooted and it worked again.

Here I was thinking I had sorted out the issue, but when I put it back on the other machine, it still doesnt work. I no longer shows as RP2 Boot, and I can see a HID Compliant Keyboard and Mouse in device manager, but no go!

I have the same issue on both, only difference is one is on windows, and one on my Proxmox host (debian).

Neither HID keyboard and mouse passthrough are working.

Thanks

mdevaev, 2023-12-16:
Could you check vendor and product ID when Pico boots in RP2 Boot mode?

deanfourie1, 2023-12-17:
Sorry, not exactly sure where to find that. I did however find this,

deanfourie1, 2023-12-17:
One question, One thing that I did not do was this. Is this required for the PiKVM v2? As I have had it working without doing the following steps.

I am again, plugging the device into multiple machine, and yet I still have NO HID keyboard and mouse control on either of the two PiKVM devices. Its almost like it only works on first boot initially after flashing the Pico.

```
The final steps[¶](https://docs.pikvm.org/pico_hid/#the-final-steps)
Connect the Pico HID to the target host using the USB cable.

If you are building PiKVM [V1](https://docs.pikvm.org/v1/), no further action with the Pico HID is required.

If you are making the Pico HID for [V2](https://docs.pikvm.org/v2/) or [V3](https://docs.pikvm.org/v3/), add the following lines to the PiKVM configuration and reboot it:

/boot/config.txt

dtoverlay=spi0-1cs
/etc/kvmd/override.yaml:

kvmd:
    hid:
        type: spi
        chip: 0
        bus: 0
        sw_cs_pin: 7
        sw_cs_per_byte: true
        reset_pin: 25
        reset_inverted: true
        reset_self: true
        power_detect_pin: 16
        power_detect_pull_down: true
```

I'm really at a loss here now. I have the USB of the Pico plugged in and the devices (Both HID keyboard and mouse show in windows) but I have no control.

deanfourie1, 2023-12-17:
Ok here's an update. I added the above to /boot/config.txt and /etc/kvmd/override.yaml and now its working but not 100%.

Sorry, I completely missed this step.

However, I still have no control over keyboard in the BIOS or PRIOR to windows booting, 

Even after windows has booted, I have to select the "Reset HID" option in PiKVM before it will start working. One I reset the HID, the mouse and keyboard work find on both, but still absolutely no BIOS support.

Any ideas?

Thanks

mdevaev, 2023-12-17:
> I completely missed this step.

Why you're making pico hid / v1 build for rpi4? Why not v2 build? You will have mass storage.

By the way, I don't see you connecting a wire with a diode to the pico. Is that true?

deanfourie1, 2023-12-17:
I'm pretty sure I've done the v2 build.

Also, no I have not connected the red wire with the diode. As it says I don't need to connect it.

mdevaev, 2023-12-17:
The V2 is build without Pico.

https://docs.pikvm.org/v2

deanfourie1, 2023-12-17:
Sorry, documentation is a little confusing. Are you saying that I dont need the Pico for the v2 build?

I see it mentions using a USB Type C Y cable, but is this for keyboard/mouse support ot just for the Flash drive access?

Is it possible to leave the Pico in place as this is already setup.

Thanks, I hope we get to the bottom of it.

mdevaev, 2023-12-18:
> I see it mentions using a USB Type C Y cable, but is this for keyboard/mouse support ot just for the Flash drive access?

Yes

> Is it possible to leave the Pico in place as this is already setup.

Yes but mass storage will not work.

Just to be sure, do you have a diode to make that connection?

deanfourie1, 2023-12-18:
Im sure I could dig one out somewhere.

deanfourie1, 2023-12-18:
Question regarding the USB, could I just solder join a USB cable together, 

ITs only D+ and D- required and GND right? Better yet, could I just use 1 usb cable and power the PI from the PC.

mdevaev, 2023-12-18:
You can use two suitable cables to make it.

[image]

deanfourie1, 2023-12-19:
Ok, I have connected the TypeC usb as per the diagram above,

Everything looks good in device manager, still no keyboard and mouse.

Any ideas?

mdevaev, 2023-12-19:
You should revert your modifications in override.yaml, or just reflash os using v2-hdmiusb-rpi4 image.

deanfourie1, 2023-12-20:
Thanks, all is working nicely now.

I do seem to have one issue with the linux Proxmox serer, video keeps cutting out randly like ever 5 seconds.

Is there anything special that needs to be done on Linux machines or any reason you can think this may be hapenning?

Seems fine on windows with same setup.

closed 2023-12-20 by mdevaev