pikvm/pikvm #1193: Audio not working using HDMI-CSI bridge

sreedharch, 2023-12-20:
Hi,
I'm streaming using HDMI-CSI bridge connection on rpi4. I notice that the stream is working fine and able to detect the audio device. 
But I see below error from ustreamer:
INFO --- => == ustreamer/audio     -- Detected host audio
INFO --- => == ustreamer/audio     -- Pipeline configured on 48000Hz; capturing ...
INFO --- => [WARN] [5558898831029643] Failed to add some remote candidates (added 0, expected 1)
INFO --- => [5558898831029643] The DTLS handshake has been completed
INFO --- => [WARN] [5558898831029643] Error resolving mDNS address (f8b55cde-4d7a-413e-a73a-229467851bdb.local): Error resolving “f8b55cde-4d7a-413e-a73a-229467851bdb.local”: Name or service not known
INFO --- => [WARN] [5558898831029643] Error resolving mDNS address (f8b55cde-4d7a-413e-a73a-229467851bdb.local): Error resolving “f8b55cde-4d7a-413e-a73a-229467851bdb.local”: Name or service not known
INFO --- => [ERR] [src/audio.c:_pcm_thread:190] == ustreamer/audio     -- Fatal: Can't capture PCM frames: Input/output error
INFO --- => == ustreamer/audio     -- Pipeline closed
INFO --- => == ustreamer/audio     -- Detected host audio
INFO --- => == ustreamer/audio     -- Pipeline configured on 48000Hz; capturing ...
INFO --- => [ERR] [src/audio.c:_pcm_thread:190] == ustreamer/audio     -- Fatal: Can't capture PCM frames: Input/output error
INFO --- => == ustreamer/audio     -- Pipeline closed
INFO --- => == ustreamer/audio     -- Detected host audio
INFO --- => == ustreamer/audio     -- Pipeline configured on 48000Hz; capturing ...
INFO --- => [ERR] [src/audio.c:_pcm_thread:190] == ustreamer/audio     -- Fatal: Can't capture PCM frames: Input/output error
 
Not sure what's missing. Any help on this part please.

mdevaev, 2023-12-20:
This is indicated in the audio documentation. The sound works only on our own hardware PiKVM V3 and V4.

sreedharch, 2023-12-20:
Ooh ok, thanks for the info @mdevaev

mdevaev, 2023-12-20:
Ur welcome.

closed 2023-12-20 by mdevaev