pikvm/pikvm #1196: I tried to control my Android phone, but nothing happened.

happymy, 2023-12-27:
I control the PC and everything is fine.
I try use USB OTG Cable to control the Android phone and there is no feedback.

mdevaev, 2023-12-27:
Try the relative mouse mode: https://docs.pikvm.org/mouse

happymy, 2023-12-28:
I tried. Still not working. Can't control my android phone. Keyboard and mouse don't work.
I observed that when I connected the USB cable, the phone could not turn on or off the USB OTG mode.

Can I set pikvm to turn off USB HOST mode? Let PIKVM emulate HID device?

mdevaev, 2023-12-28:
PiKVM is emulating HID device by default. Presumably there are problems in Android.

happymy, 2023-12-30:
I cut the red power wire inside the USB cable.
Nothing has changed.

I guess the both USB work in HOST mode. Need disable pikvm USB host mode？

mdevaev, 2023-12-30:
PiKVM never works in the host mode, only in the device mode. Probably your phone is a device too.

mdevaev, 2023-12-30:
Which connector do you have on the phone? Micro USB or Type C?

happymy, 2023-12-31:
Type-C.
That's weird. The mobile phone can turn on and off USB OTG.

happymy, 2024-01-02:
That is work！ I try add a usb HUB to usb otg cable.

mdevaev, 2024-01-09:
So, the problem solved?

happymy, 2024-01-10:
yes. But why?

happymy, 2024-01-11:
Why add a usb HUB to usb otg cable first?

mdevaev, 2024-01-11:
I suppose this helped the phone determine the type of connection (host/device).

closed 2024-01-13 by mdevaev