pikvm/pikvm #1197: Keyboard not working but mouse working. Win11 21H2 22000.348

wzqvip, 2023-12-27:
**Describe the bug**
The keyboard not working, including the shortcuts button and paste. Mouse and video are working well.
Destination machine is running Win11 Pro 21H2 22000.348.

Destination: 
 - OS: Windows11 Pro 21H2 22000.348
 - Gigabyte Z690. I9-12900K.

**Source Desktop (please complete the following information):**
 - Browser Chrome/Edge
 - Version Newest.

**PiKVM info:**
  - Rpi zero 2w
  - v2-hdmi-zero2w-latest.img.xz (download on Dec 12, 2023)
  - CSI bridge
  - KVMD version: `kvmd 3.276-1
                              kvmd-fan 0.26-1
                              kvmd-oled 0.26-1
                              kvmd-platform-v2-hdmi-zero2w 3.276-1
                              kvmd-webterm 0.44-1`
  - uStreamer version: `ustreamer 5.45-1`
  - Linux kernel: `Linux pikvm 6.1.61-1-rpi-ARCH #1 SMP Fri Nov  3 20:48:52 MSK 2023 armv7l GNU/Linux`

**Additional context**
I tried to unplug the mouse/keyboard on the destination machine, not working.
Both the _mouse_ and the _keyboard_ on the physical machine works, 
mouse and keyboard on KVM works on other machines, only this machine does not work. On both the login screen and the desktop, the keyboard does not function at all. Weird.

arch1mede, 2023-12-28:
Hi, I would suggest joining [Discord](https://discord.gg/bpmXfz5) to get further assistance.

I would try Chrome Incognito mode and see if the behavior changes, otherwise I will say that this is possibly due to the Target OS.

wzqvip, 2024-01-02:
That's weird. I've tried but still not working. 
I've used it on plenty of machines, and this is the only machine that "keyboard" not works.

arch1mede, 2024-01-12:
I have seen some strange things related to [EDID](https://docs.pikvm.org/edid/), I would look into that first to see if you can get the behavior to change.

Otherwise

Logically speaking, if your pikvm works on other systems but not this one, then its likely a system issue. There isn't much that can be suggested to diagnose the issue other than reinstalling the OS.

arch1mede, 2024-02-16:
Hi, did you get this working? If not, we would ask that you join us on Discord to troubleshoot further. While the EDID "might" play into your issue, this could also be some issue related to the OS or USB port, we would recommend if you could try this on another system, if this works on another system and does not work on this system, then I would logically assume that either the USB port is not working correctly or the OS needs to be reinstalled. We cant know what the actual issue is till you provide us additional info via Discord.

If you have resolved this issue, please mark this as resolved.

wzqvip, 2024-02-16:
> Hi, did you get this working? If not, we would ask that you join us on Discord to troubleshoot further. While the EDID "might" play into your issue, this could also be some issue related to the OS [...]

I'll join the discord. Mouse and keyboard can work in BIOS and before entering system. The keyboard won't work only in system. It's really weird, and I won't reinstall the system for some reason. I can use the screen keyboard with mouse to do simple tasks.

closed 2024-03-16 by mdevaev