pikvm/pikvm #1198: Keyboard not working in Web Terminal since update

Leseratte10, 2023-12-28:
**Describe the bug**

Flashed PiKVM image `v3-hdmi-rpi4-latest.img.xz` downloaded today, file timestamp from December 6th. Web terminal works. Updated OS, web terminal no longer works. It opens and I can see the text, but all keyboard input is ignored. 

Also, interestingly, prior to the update the PiKVM had IPv4 and IPv6 addresses. After the update it is only reachable over IPv6 and doesn't have a local IPv4 anymore.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Flash image from December 6th
2. Open Web Terminal, check that it works
3. run "rw", run "pacman -Syu", run "reboot"
4. Web Terminal broken & Pi only reachable over IPv6

**Expected behavior**
Expect Pi to still have both IPv4 and IPv6, and web terminal to work.

**Desktop:**
 - OS: Ubuntu 20.04
 - Browser: Tested Chrome 116.0.5845.140 and Firefox 117.0

**PiKVM info:**
  - Raspberry Pi board version: CM4
  - PiKVM platform: v3-hdmi-rpi4
  - Video capture type: CSI bridge
  - KVMD version before update (working):

```
kvmd 3.284-1
kvmd-fan 0.26-1
kvmd-oled 0.26-1
kvmd-platform-v3-hdmi-rpi4 3.284-1
kvmd-webterm 0.44-1
```  

  - KVMD version after update (broken):
 
```
kvmd 3.288-1
kvmd-fan 0.26-1
kvmd-oled 0.26-1
kvmd-platform-v3-hdmi-rpi4 3.288-1
kvmd-webterm 0.47-1
```
  - uStreamer version: ustreamer 5.46-1
  - Linux kernel: `Linux pikvm 6.1.61-1-rpi-ARCH #1 SMP Fri Nov  3 20:48:52 MSK 2023 armv7l GNU/Linux` (didn't change during update)

If you need any additional information let me know.

Cebrain, 2023-12-28:
Can confirm the problem.

I flashed the "v2-hdmiusb-rpi4-latest.img" from today. 
After the update AND reboot, i cant type in the terminal anymore.

SSH and WebGUI works fine.

Leseratte10, 2023-12-28:
Looks like the update to ttyd 1.7.4 broke it - the [upstream changelog](https://github.com/tsl0922/ttyd/releases/tag/1.7.4) says:

    Breaking Changes
    The web terminal is readonly by default now, to make it writable, use -W option.

So I guess some PiKVM code needs to be updated to include that switch.

mdevaev, 2023-12-29:
I'll build the new package with fix, thank you.

mdevaev, 2023-12-29:
Fixed in new kvmd-webterm

closed 2023-12-29 by mdevaev