pikvm/pikvm #1199: After update: 500 Internal Server Error

tomatotoast, 2023-12-30:
**Describe the bug**
After running:

[root@pikvm ~]# pacman -Syu
:: Synchronizing package databases...
 core is up to date
 extra                   9.0 MiB  6.09 MiB/s 00:01 [######################] 100%
 community is up to date
 alarm is up to date
 aur is up to date
 pikvm is up to date
:: Starting full system upgrade...
warning: python-ajsonrpc: local (1.2.0-4) is newer than extra (1.2.0-3)
resolving dependencies...
looking for conflicting packages...

Packages (2) libgpiod-2.0-1  python-fastjsonschema-2.19.1-1

Total Download Size:   0.16 MiB
Total Installed Size:  0.83 MiB
Net Upgrade Size:      0.23 MiB

:: Proceed with installation? [Y/n] 
:: Retrieving packages...
 libgpiod-2.0-1-a...   120.8 KiB   575 KiB/s 00:00 [######################] 100%
 python-fastjsons...    47.1 KiB  1571 KiB/s 00:00 [######################] 100%
 Total (2/2)           167.9 KiB   420 KiB/s 00:00 [######################] 100%
(2/2) checking keys in keyring                     [######################] 100%
(2/2) checking package integrity                   [######################] 100%
(2/2) loading package files                        [######################] 100%
(2/2) checking for file conflicts                  [######################] 100%
(2/2) checking available disk space                [######################] 100%
:: Processing package changes...
(1/2) upgrading libgpiod                           [######################] 100%
(2/2) upgrading python-fastjsonschema              [######################] 100%
:: Running post-transaction hooks...
(1/1) Arming ConditionNeedsUpdate...

i get a 500 error on the webinterface.

**To Reproduce**
described above

**Expected behavior**
This should not happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**PiKVM info:**
  - Raspberry Pi board version  RPi 4
  - PiKVM platform Geekworm KVM-A8 PCIe
  - Video capture type CSI bridge
  - KVMD version: kvmd 3.288-1
                             kvmd-fan 0.26-1
                             kvmd-oled 0.26-1
                             kvmd-platform-v3-hdmi-rpi4 3.288-1
                             kvmd-webterm 0.48-1

  - uStreamer version: ustreamer 5.46-1
  - Linux kernel: 6.1.61-1-rpi-ARCH

tomatotoast, 2023-12-30:
Simple fix by rolling back the updated packages and reboot:

```
pacman -U file:///var/cache/pacman/pkg/python-fastjsonschema-2.19.0-1-any.pkg.tar.xz
pacman -U file:///var/cache/pacman/pkg/libgpiod-1.6.4-2-armv7h.pkg.tar.xz
```

andreaseckmayr, 2023-12-30:
Thanks!

Second command should be `pacman -U file:///var/cache/pacman/pkg/libgpiod1-1.6.4-2-armv7h.pkg.tar.xz`.

fiveways, 2023-12-30:
I have a similar error. kvmd could not start due to plugin not found ugpio/gpio

rolling back to libgpiod-1.6.4-2-armv7h.pkg.tar.xz allowed kvmd to start

mdevaev, 2023-12-30:
Use this command to update and everything will be ok: `curl https://files.pikvm.org/update-os.sh | bash`

closed 2023-12-30 by mdevaev