pikvm/pikvm #1202: Controlling iOS/iPad OS device does not send Left Clicks (but sends keyboard, mouse move and right click)

alcor, 2024-01-03:
**Describe the bug**
When piKVM is connected to an iOS device, specifically an iPad, it sends Keyboard, mouse move and Right click, but not Left click

**To Reproduce**
Connect a pikvm device to an ipad with an adapter, connect to it remotely.

**Expected behavior**
Left clicks should be seen by the OS 

**Screenshots**
Example of desktop view
[image]

**Desktop (please complete the following information):**
 - OS: macOS Ventura
 - Browser: Arc (Chrome)
 - Version: 120

**PiKVM info:**
  - Raspberry Pi board version: RPi 4 and Zero 2 both exhibit this
  - PiKVM platform: V2 (HDMI)
  - Video capture type: Both USB and CSD
  - KVMD version: 
      kvmd 3.290-1
      kvmd-fan 0.27-1
      kvmd-oled 0.26-1
      kvmd-platform-v2-hdmi-zero2w 3.290-1
      kvmd-webterm 0.48-1

  - uStreamer version: `pacman -Q | grep ustreamer`
      ustreamer 5.46-3
  - Linux kernel: `uname -a`
       Linux pikvm 6.1.61-1-rpi-ARCH #1 SMP Fri Nov  3 20:48:52 MSK 2023 armv7l GNU/Linux

**Additional context**
I've tried this with a couple different piKVM devices, and 2 different iPads (11 pro and mini) - in both cases, keyboard events, mouse move events and right clicks work, but left clicks are ignored. Thanks for considering!

alcor, 2024-01-03:
May be slightly related: because the iPad screen is not exactly 16:9, the cursor position varies slightly as you get to the screen edges.

mdevaev, 2024-01-05:
I'll check it.

mdevaev, 2024-01-07:
I've tested it. You need to use [the relative mouse mode](https://docs.pikvm.org/mouse) because the iOS absolute mouse driver is buggy. I mean, this is a problem from Apple side, we can't fix it. But the relative mouse works well.

closed 2024-01-07 by mdevaev

ewspikvm, 2025-10-31:
You may also need to enable the Accessibility option called AssistiveTouch as this seems to change the ability of iPadOS and iOS to support some additional input device types. Just turning it on fixes left click (at least when wired), through Bluetooth it might not hurt either. There are still glitches, but as mdevaev mentioned, those are Apple quirks and not something that can be fixed on the PiKVM side.