pikvm/pikvm #1203: Updating broken on keyring

TheKigen, 2024-01-04:
**Describe the bug**
When updating my PiKVM v4 the process breaks on the keyring check.

**To Reproduce**
Steps to reproduce the behavior, like:
1. SSH the PiKVM
2. Run `curl https://files.pikvm.org/update-os.sh | bash`

**Expected behavior**
It is expected to update successfully.

**Logs**
```
[root@pikvm ~]# curl https://files.pikvm.org/update-os.sh | bash
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   953  100   953    0     0   2640      0 --:--:-- --:--:-- --:--:--  2654
:: Synchronizing package databases...
 core                                                                                              243.1 KiB   507 KiB/s 00:00 [#############################################################################] 100%
 extra                                                                                               9.0 MiB  23.0 MiB/s 00:00 [#############################################################################] 100%
 community                                                                                          45.0   B   642   B/s 00:00 [#############################################################################] 100%
 alarm                                                                                              95.6 KiB  1366 KiB/s 00:00 [#############################################################################] 100%
 aur                                                                                                 9.3 KiB   133 KiB/s 00:00 [#############################################################################] 100%
 pikvm                                                                                              46.2 KiB  79.6 KiB/s 00:01 [#############################################################################] 100%
warning: raspberrypi-utils-20231221-2 is up to date -- reinstalling
warning: ustreamer-5.46-3 is up to date -- reinstalling
warning: pikvm-os-raspberrypi-0.5-1 is up to date -- reinstalling
warning: libgpiod1-1.6.4-4 is up to date -- reinstalling
warning: kvmd-fan-0.27-1 is up to date -- reinstalling
warning: kvmd-3.290-1 is up to date -- reinstalling
warning: kvmd-platform-v4plus-hdmi-rpi4-3.290-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (7) kvmd-3.290-1  kvmd-fan-0.27-1  kvmd-platform-v4plus-hdmi-rpi4-3.290-1  libgpiod1-1.6.4-4  pikvm-os-raspberrypi-0.5-1  raspberrypi-utils-20231221-2  ustreamer-5.46-3

Total Installed Size:  5.42 MiB
Net Upgrade Size:      0.00 MiB

:: Proceed with installation? [Y/n]
(7/7) checking keys in keyring                                                                                                 [#############################################################################] 100%
downloading required keys...
:: Import PGP key 912C773ABBD1B584, "Unknown Packager"? [Y/n]
error: key "912C773ABBD1B584" could not be imported
error: required key missing from keyring
error: failed to commit transaction (unexpected error)
Errors occurred, no packages were upgraded.
```

arch1mede, 2024-01-05:
> Edit: To note, I have tried accepting the key since it is the configured signing key of the PiKVM project maintainer, but every attempt fails.
Hi, can you provide what commands you ran and what the results of those commands were?

TheKigen, 2024-01-05:
Running a normal `pacman -Syu` it'll popup that same question.  Inputting `Y` does nothing.
```
downloading required keys...
:: Import PGP key 912C773ABBD1B584, "Unknown Packager"? [Y/n] Y
error: key "912C773ABBD1B584" could not be imported
error: required key missing from keyring
error: failed to commit transaction (unexpected error)
Errors occurred, no packages were upgraded.
```

Other attempts at retrieving the key do not work.  Honestly I'm close to have to just reflash this PiKVM v4 Plus with a fresh image.

```
[root@pikvm ~]# pacman-key --recv-keys 912C773ABBD1B584
gpg: [don't know]: invalid packet (ctb=00)
gpg: keydb_get_keyblock failed: Value not found
gpg: [don't know]: invalid packet (ctb=00)
gpg: /etc/pacman.d/gnupg/pubring.gpg: copy to '/etc/pacman.d/gnupg/pubring.gpg.tmp' failed: Invalid packet
gpg: error writing keyring '/etc/pacman.d/gnupg/pubring.gpg': Invalid packet
gpg: error reading '[stream]': Invalid packet
gpg: Total number processed: 0
==> ERROR: Remote key not fetched correctly from keyserver.
[root@pikvm ~]# pacman-key --keyserver hkps://hkps.pool.sks-keyservers.net -r 912C773ABBD1B584
gpg: keyserver receive failed: Try again later
==> ERROR: Remote key not fetched correctly from keyserver.
[root@pikvm ~]# pacman-key --keyserver hkps://keyserver.ubuntu.com:443 -r 912C773ABBD1B584
gpg: [don't know]: invalid packet (ctb=00)
gpg: keydb_get_keyblock failed: Value not found
gpg: [don't know]: invalid packet (ctb=00)
gpg: /etc/pacman.d/gnupg/pubring.gpg: copy to '/etc/pacman.d/gnupg/pubring.gpg.tmp' failed: Invalid packet
gpg: error writing keyring '/etc/pacman.d/gnupg/pubring.gpg': Invalid packet
gpg: error reading '[stream]': Invalid packet
gpg: Total number processed: 0
==> ERROR: Remote key not fetched correctly from keyserver.
[root@pikvm ~]# pacman-key --keyserver hkps://hkps.pool.sks-keyservers.net:443 -r 912C773ABBD1B584
gpg: keyserver receive failed: No name
==> ERROR: Remote key not fetched correctly from keyserver.
[root@pikvm ~]# pacman-key --keyserver hkps://keyserver.ubuntu.com -r 912C773ABBD1B584
gpg: [don't know]: invalid packet (ctb=00)
gpg: keydb_get_keyblock failed: Value not found
gpg: [don't know]: invalid packet (ctb=00)
gpg: /etc/pacman.d/gnupg/pubring.gpg: copy to '/etc/pacman.d/gnupg/pubring.gpg.tmp' failed: Invalid packet
gpg: error writing keyring '/etc/pacman.d/gnupg/pubring.gpg': Invalid packet
gpg: error reading '[stream]': Invalid packet
gpg: Total number processed: 0
==> ERROR: Remote key not fetched correctly from keyserver.
```

jacob-wm, 2024-01-06:
I can confirm the exact same thing for myself. I updated all the other system packages, but down to the last three KVM-related that fail on that very key.

I also was unable to override this.

mdevaev, 2024-01-06:
Anyone who have this bug, please contact me via PM in discord: https://discord.gg/bpmXfz5

mdevaev, 2024-01-06:
Please just retry:
```
rw
pacman-key --keyserver hkps://keyserver.ubuntu.com -r 912C773ABBD1B584
```

TheKigen, 2024-01-07:
```
[root@pikvm ~]# rw
[root@pikvm ~]# pacman-key --keyserver hkps://keyserver.ubuntu.com -r 912C773ABBD1B584
gpg: [don't know]: invalid packet (ctb=00)
gpg: keydb_get_keyblock failed: Value not found
gpg: [don't know]: invalid packet (ctb=00)
gpg: /etc/pacman.d/gnupg/pubring.gpg: copy to '/etc/pacman.d/gnupg/pubring.gpg.tmp' failed: Invalid packet
gpg: error writing keyring '/etc/pacman.d/gnupg/pubring.gpg': Invalid packet
gpg: error reading '[stream]': Invalid packet
gpg: Total number processed: 0
==> ERROR: Remote key not fetched correctly from keyserver.
```

TheKigen, 2024-01-07:
So I have a second PiKVM v4 Plus that is upgrading just fine.  I have found that there appears to be something that caused `/etc/pacman.d/gnupg/pubring.gpg` to become corrupted.  Copying the file from the known good PiKVM fixed the issue.

mdevaev, 2024-01-07:
Also can help:
```
rw
pacman-key --init
pacman-key --populate
pacman-key --keyserver hkps://keyserver.ubuntu.com -r 912C773ABBD1B584
```

mdevaev, 2024-01-10:
Ok I'm closing it since we have a workaround. The problem is not caused by our actions, it is something that breaks nondeterministic.

closed 2024-01-10 by mdevaev

ajorpheus, 2024-03-12:
> Also can help:
> 
> ```
> rw
> pacman-key --init
> pacman-key --populate
> pacman-key --keyserver hkps://keyserver.ubuntu.com -r 912C773ABBD1B584
> ```

In addition to the above, I had to do the following:

```
sudo pacman-key --recv-keys 912C773ABBD1B584
sudo pacman-key --lsign-key 912C773ABBD1B584
```

mdevaev, 2024-03-21:
@ajorpheus Added to my answer, thank you.

47k, 2024-04-23:
For me it's not working:

> [root@pikvm ~]# pacman-key --init
> gpg: checking the trustdb
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keyring_get_keyblock: read error: Invalid packet
> gpg: keyring_get_keyblock failed: Invalid keyring
> gpg: failed to rebuild keyring cache: Invalid keyring
> gpg: marginals needed: 3  completes needed: 1  trust model: pgp
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keyring_get_keyblock: read error: Invalid packet
> gpg: keydb_get_keyblock failed: Invalid keyring
> gpg: validate_key_list failed
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keyring_get_keyblock: read error: Invalid packet
> gpg: keydb_get_keyblock failed: Invalid keyring
> [root@pikvm ~]# pacman-key --populate
> gpg: checking the trustdb
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keyring_get_keyblock: read error: Invalid packet
> gpg: keyring_get_keyblock failed: Invalid keyring
> gpg: failed to rebuild keyring cache: Invalid keyring
> gpg: marginals needed: 3  completes needed: 1  trust model: pgp
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keyring_get_keyblock: read error: Invalid packet
> gpg: keydb_get_keyblock failed: Invalid keyring
> gpg: validate_key_list failed
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keyring_get_keyblock: read error: Invalid packet
> gpg: keydb_get_keyblock failed: Invalid keyring
> ==> Appending keys from archlinux.gpg...
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keydb_search failed: Invalid packet
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keydb_search failed: Invalid packet
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keydb_search failed: Invalid packet
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keydb_search failed: Invalid packet
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keydb_search failed: Invalid packet
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keydb_search failed: Invalid packet
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keydb_search failed: Invalid packet
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keydb_search failed: Invalid packet
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keydb_search failed: Invalid packet
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keydb_search failed: Invalid packet
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keydb_search failed: Invalid packet
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keydb_search failed: Invalid packet
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keydb_search failed: Invalid packet
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keydb_search failed: Invalid packet
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keydb_search failed: Invalid packet
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keyring_get_keyblock: read error: Invalid packet
> gpg: keydb_get_keyblock failed: Invalid keyring
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: /etc/pacman.d/gnupg/pubring.gpg: copy to '/etc/pacman.d/gnupg/pubring.gpg.tmp' failed: Invalid packet
> gpg: error writing keyring '/etc/pacman.d/gnupg/pubring.gpg': Invalid packet
> gpg: error reading '/usr/share/pacman/keyrings/archlinux.gpg': Invalid packet
> gpg: no valid OpenPGP data found.
> gpg: import from '/usr/share/pacman/keyrings/archlinux.gpg' failed: Invalid packet
[... 582 lines cut ...]
> gpg: keyring_get_keyblock: read error: Invalid packet
> gpg: keyring_get_keyblock failed: Invalid keyring
> gpg: failed to rebuild keyring cache: Invalid keyring
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keyring_get_keyblock: read error: Invalid packet
> gpg: keydb_get_keyblock failed: Invalid keyring
> gpg: validate_key_list failed
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keydb_search failed: Invalid packet
> gpg: error reading key: Invalid packet
> ==> Locally signing trusted keys in keyring...
> ==> ERROR: 75BD80E4D834509F6E740257B1B73B02CC52A02A could not be locally signed.
> ==> ERROR: 2AC0A42EFB0B5CBC7A0402ED4DC95B6D7BE9892E could not be locally signed.
> ==> ERROR: 69E6471E3AE065297529832E6BA0F5A2037F4F41 could not be locally signed.
> ==> ERROR: D8AFDDA07A5B6EDFA7D8CCDAD6D055F927843F1C could not be locally signed.
> ==> ERROR: 91FFE0700E80619CEB73235CA88E23E377514E00 could not be locally signed.
> [root@pikvm ~]# pacman-key --keyserver hkps://keyserver.ubuntu.com -r 912C773ABBD1B584
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: keydb_get_keyblock failed: Value not found
> gpg: [don't know]: invalid packet (ctb=00)
> gpg: /etc/pacman.d/gnupg/pubring.gpg: copy to '/etc/pacman.d/gnupg/pubring.gpg.tmp' failed: Invalid packet
> gpg: error writing keyring '/etc/pacman.d/gnupg/pubring.gpg': Invalid packet
> gpg: error reading '[stream]': Invalid packet
> gpg: Total number processed: 0
> ==> ERROR: Remote key not fetched correctly from keyserver.

mdevaev, 2024-04-23:
Please reflash it.

TheKigen, 2024-04-23:
I'm pretty sure you need to `rm /etc/pacman.d/gnupg/pubring.gpg` before running the `pacman-key --init` command.  Since the `pacman-key` command always tries to read the pubring.

47k, 2024-04-24:
Thank you - I had a lof of other troubles with some keys after hat command but I could fix it. Seems that my device is up2date :)

jomarx, 2024-06-04:
I had to do this to make it work in my case. 

[https://wiki.archlinux.org/title/Pacman/Package_signing#Resetting_all_the_keys](url)

nrasmunsen, 2025-01-24:
I want to leave my experience here in case it helps somebody else.

I had the same issue with the keyring being broken but the resolution steps were not working for me, I got stuck when trying to pull the key from the server. It wouldn't resolve and would throw back the error: Unable to fetch GPG key from keyserver.

It turns out that it was my Tailscale configuration getting in the way, the nameserver was not resolving.

What I did to resolve it was:
1. Turn off tailscale (tailscale down).
2. Edit /etc/resolv.conf file to remove the DNS from tailscale and add 8.8.8.8 instead.
3. Tried to run the steps mentioned above to reset the keyring. Now they run as expected.
4. Run the pikvm-update command and now it worked.

Thanks for the guidance here and for putting me in the right path, my pikvm is thousands of miles away and reflashing was an expensive option involving multiple flights.