pikvm/pikvm #1204: kvmd session timeout

cconkrig, 2024-01-06:
**Describe the bug**
HTTP/Web authentication seems to never timeout/expire unless you restart kvmd service or manually logout. If you restart the kvmd service, then auth is cleared and users are forced to login again. This is a security risk if someone forgets to logout manually.

**To Reproduce**
1.) Login to the web gui.
2.) Open the KVM control to display the remote computer.
3.) Close the browser (do not logout as a lot of users will do).
4.) Re-open the browser (even days/weeks? later) and you are still logged in.

**Expected behavior**
You should be re-prompted to login.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser: Chrome
 - Version: 120.0.6099.130 (64-bit)
 - VNC client (if used): N/A

**PiKVM info:**
  - Raspberry Pi board version: RPi 4
  - PiKVM platform: v3-hdmi
  - Video capture type: CSI bridge
  - KVMD version: 
        kvmd 3.287-1
        kvmd-fan 0.26-1
        kvmd-oled 0.26-1
        kvmd-platform-v3-hdmi-rpi4 3.287-1
        kvmd-webterm 0.47-1
  - uStreamer version: ustreamer 5.45-1
  - Linux kernel: Linux pikvm 6.1.61-1-rpi-ARCH #1 SMP Fri Nov  3 20:48:52 MSK 2023 armv7l GNU/Linu

mdevaev, 2024-01-06:
This is implemented intentionally so as not to bother the user. Personally, my eyelid twitches every time sites ask me to relogin without any meaning. Are you sure this is necessary?

cconkrig, 2024-01-07:
Yes. It's a major security risk if a user doesn't remember to log out and walks away. It should be a flag or be configurable in the yaml.  Why have OTP if you are just always logged in?

mdevaev, 2024-01-07:
I assume that if a user uses a computer with public access, then he should lock the screen when he leaves or log out on his own. Otherwise, there is no safe time interval, because even five minutes of physical access is enough to steal cookies or do something else. In short, a timeout does not solve the security problem in any way.

mdevaev, 2024-01-10:
What interval do you consider acceptable?

mdevaev, 2024-01-28:
Sup?

safesploit, 2024-08-04:
This `session_timeout` should be defined in the config file, such that it can be chosen by the admin. 
Otherwise, have `session_timeout = 0` which allow the user to remain logged in forever.

I think the issue @cconkrig was raising more focused on not having the choice to decide.

KingRat0011, 2024-08-06:
I do think this is needed, it is a security risk for me when someone leaves the session open, so I upvote a timeout please, like a auto logout

adamoutler, 2025-01-25:
I'm using Chromium under Ubuntu and I was logged out when I closed the browser.  

[image]

mdevaev, 2025-02-09:
I added the login option to specify the time when the session will expire.

* By default, an infinite session is used, but if the user switch the time selector, the browser will remember it.
* It is possible to set the global config value of maximum time in accordance with your security policy (is not set by default = infinite). User without root privileges can't change it.
* If the global limit is set, but the user wants a longer session, he won't get it.
* If the session is expired, it does not affect other sessions of the same user.
* Clicking `Logout` leads to the logout of all sessions of this user, as it was before.

The default settings are selected for backward compatibility reasons. This will be available in the next release.

Note the expiration of the session invalidates the authorization token. This means that no new connections will be established, but existing long-lived connections (websockets for hid, video) will not be terminated. It seems that usually no one cares about similar problems, and closing a browser tab by a decent user solves it. However, it seems to me a good idea to plan a way to immediately terminate all connections when the token was expired. Like, time's up, get out :) 

[image]

mdevaev, 2025-02-10:
Released: https://docs.pikvm.org/auth/#session-expiration

closed 2025-02-10 by mdevaev