pikvm/pikvm #1206: Keyboard events are always consumed when the screen is in full screen

yurivict, 2024-01-08:
**Is your feature request related to a problem? Please describe.**

Please allow the user to toggle the connection option 'Grab all keyboard events' when the screen is in a fullscreen mode.
This way the user would be able to "glance" into the connection with Alt-Tab and go away with another Alt-Tab, and not get stuck in that screen.

Remmina does this right: It allows the user to toggle the 'Grab all keyboard events' option with the button and the keyboard shortcut.

**Describe the solution you'd like**
See above.

**Describe alternatives you've considered**
n/a

**Additional context**
This feature is very useful when the user shares time between different systems and needs to be able to easily go back and forth.

Kolyunya, 2025-09-24:
I'm using Chrome in the full-screen `application` mode on MacOS, and unfortunately, `Command+Tab` triggers window switch on the host machine instead of the remote one. Is there a way to fix that behaviour?

mdevaev, 2025-10-26:
> Is there a way to cycle through windows of the remote machine using a standard keyboard shortcut?

No. It's a big pain becuse some keys on macos cannot be intercepted in the browser at all.

mdevaev, 2025-11-16:
@Kolyunya here is the workaround: https://docs.pikvm.org/shortcuts/