pikvm/pikvm #1207: Please add the ability to have user defined text on the KVM screen

cwkreimer, 2024-01-08:
It would be extremely helpful if text could be added to the top of the two windows and on the sides of the KVM screens. This would be very helpful for management and maintenance of a large number of computers / servers that have different requirements to get into the BIOS and boot menus.  Some PCs use F2 to enter the BIOS screen.  Others use “ESC”.    We have a small fleet of PiKVM v4 Plus and this would be a large help to heep things straight for the IT guys.

It would make management and maintenance of servers/PCs a lot simpler to be able to put HTML into a file and have it displayed on the screen in a manner that calls your attention to it.  If possible, with CSS – but being able to put the text in red would be great if that is all that is possible.

This could be data in a .yaml file with 4 sections to cover the four arrows in the attached  screen shot

[image]

mdevaev, 2024-11-05:
I've added left and right columns configurable via meta.yaml like this:

```yaml
server:
    host: localhost.localdomain

kvm: {}

tips:
    left: test left
    right: "Lorem Ipsum is simply dummy text of the printing and typesetting industry.

Lorem Ipsum has been the industry's standard dummy text ever since the 1500s, when an unknown printer took a galley of type and scrambled it to make a type specimen book. It has survived not only five centuries, but also the leap into electronic typesetting, remaining essentially unchanged. It was popularised in the 1960s with the release of Letraset sheets containing Lorem Ipsum passages, and more recently with desktop publishing software like Aldus PageMaker including versions of Lorem Ipsum."
```

[image]

You can also change a visual style via `/etc/kvmd/web.css`:
```css
#kvmd-meta-tips-right {
        color: red;
}
```

I did not make the text in the navbar and window header because I want to leave this space reserved for future changes.

It will be available in the next release.

closed 2024-11-05 by mdevaev

CWwwKr, 2024-11-05:
Great thank you! Do you have documentation for this?

Closed #1207<https://github.com/pikvm/pikvm/issues/1207> as completed.

CWwwKr, 2024-11-05:
Sorry. I didn't see this email. Thank you very much for your help!

image.png (view on web)<https://github.com/user-attachments/assets/7ef0c496-e74d-4821-881b-6e2acd58eaab>

mdevaev, 2024-11-05:
:ok_hand:

CWwwKr, 2024-11-05:
Again thank you very much! Do you know when your next update is going to be?

👌

mdevaev, 2024-11-24:
Released now.

CWwwKr, 2024-11-24:
Thanks!  I will update one of KVMs and try it out. 🙂