pikvm/pikvm #1208: When VNC keymap /usr/share/kvmd/keymaps/en-us is used - the ':' key is translated to '^'

yurivict, 2024-01-08:
**Describe the bug**
Shift-; should enter the ':' character, but instead it enters the '^' character.

/usr/share/kvmd/keymaps/en-gb doesn't have such problem.

mdevaev, 2024-01-08:
Which layout do you have on client and dest PCs?

yurivict, 2024-01-08:
It's US English on Windows 11 Enterprise.

yurivict, 2024-01-08:
I changed to the en-gb keymap. But with this keymap there is also a problem: the ~ key (Shift-`) is entered as ^ from the VNC client.

Web interface is ok.

mdevaev, 2024-01-10:
You use Remmina, right? Using keymaps is actually a deprecated feature.

The fact is that VNC is fundamentally incompatible with hardware keyboard emulation: it transmits keys in the form of X11 codes. The reverse conversion from X11 to physical codes is inaccurate.

In order for VNC to be used with hardware keys, QEMU has developed a [protocol extension](https://github.com/rfbproto/rfbproto/blob/master/rfbproto.rst#qemu-extended-key-event-message) that allows physical keys to be transferred. Almost all modern clients support this, except Remmina. If you have the time and skill, you can make them a patch, they are willing to accept the changes. A couple of years ago I wanted to do this, the implementation is not too complicated, but I did not have time to do it.

yurivict, 2024-01-10:
Yes, I use Remmina.

mdevaev, 2024-01-10:
What about patching?

Btw try `ru` keymap.

yurivict, 2024-01-11:
Yes, I will prepare the PR for Remmina.

mdevaev, 2024-01-11:
Got it. Thank you.

mdevaev, 2024-01-28:
Okay, I'm closing this as you're solving this on the client side. Let me know about the progress, please.

closed 2024-01-28 by mdevaev