pikvm/pikvm #1209: Upgrade fails with command curl https://files.pikvm.org/update-os.sh | bash

HB9VQQ, 2024-01-09:
Currently on 
[image]

Tried to upgrade with "https://files.pikvm.org/update-os.sh | bash" but it failed

[root@pikvm ~]# rw
[root@pikvm ~]# curl https://files.pikvm.org/update-os.sh | bash
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   953  100   953    0     0   3427      0 --:--:-- --:--:-- --:--:--  3428
:: Synchronizing package databases...
 core                                                                                                243.9 KiB   581 KiB/s 00:00 [##############################################################################] 100%
 extra                                                                                                 9.0 MiB  19.5 MiB/s 00:00 [##############################################################################] 100%
 community                                                                                            45.0   B  1500   B/s 00:00 [##############################################################################] 100%
 alarm                                                                                                95.7 KiB   233 KiB/s 00:00 [##############################################################################] 100%
 aur                                                                                                   9.3 KiB   233 KiB/s 00:00 [##############################################################################] 100%
 pikvm                                                                                                46.1 KiB   171 KiB/s 00:00 [##############################################################################] 100%
warning: raspberrypi-utils-20231221-2 is up to date -- reinstalling
warning: ustreamer-5.48-1 is up to date -- reinstalling
warning: pikvm-os-raspberrypi-0.5-1 is up to date -- reinstalling
warning: libgpiod1-1.6.4-4 is up to date -- reinstalling
warning: kvmd-fan-0.28-1 is up to date -- reinstalling
warning: kvmd-3.291-1 is up to date -- reinstalling
warning: kvmd-platform-v4mini-hdmi-rpi4-3.291-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (7) kvmd-3.291-1  kvmd-fan-0.28-1  kvmd-platform-v4mini-hdmi-rpi4-3.291-1  libgpiod1-1.6.4-4  pikvm-os-raspberrypi-0.5-1  raspberrypi-utils-20231221-2  ustreamer-5.48-1

Total Installed Size:  5.42 MiB
Net Upgrade Size:      0.00 MiB

:: Proceed with installation? [Y/n] 
(7/7) checking keys in keyring                                                                                                   [##############################################################################] 100%
(7/7) checking package integrity                                                                                                 [##############################################################################] 100%
(7/7) loading package files                                                                                                      [##############################################################################] 100%
(7/7) checking for file conflicts                                                                                                [##############################################################################] 100%
(7/7) checking available disk space                                                                                              [##############################################################################] 100%
warning: could not get file information for usr/bin/vcgencmd
:: Processing package changes...
(1/7) reinstalling raspberrypi-utils                                                                                             [##############################################################################] 100%
(2/7) reinstalling libgpiod1                                                                                                     [##############################################################################] 100%
(3/7) reinstalling ustreamer                                                                                                     [##############################################################################] 100%
(4/7) reinstalling pikvm-os-raspberrypi                                                                                          [##############################################################################] 100%
(5/7) reinstalling kvmd-fan                                                                                                      [##############################################################################] 100%
(6/7) reinstalling kvmd                                                                                                          [##############################################################################] 100%
==> Ensuring KVMD users and groups ...
==> Patching configs ...
(7/7) reinstalling kvmd-platform-v4mini-hdmi-rpi4                                                                                [##############################################################################] 100%
:: Running post-transaction hooks...
(1/5) Creating system user accounts...
(2/5) Reloading system manager configuration...
(3/5) Creating temporary files...
(4/5) Reloading device manager configuration...
(5/5) Arming ConditionNeedsUpdate...
:: Starting full system upgrade...
resolving dependencies...
warning: cannot resolve "dbus-units", a dependency of "systemd"
warning: cannot resolve "dbus-units", a dependency of "systemd"
warning: cannot resolve "dbus-units", a dependency of "systemd"
:: The following package cannot be upgraded due to unresolvable dependencies:
      systemd

:: Do you want to skip the above package for this upgrade? [y/N] 
error: failed to prepare transaction (could not satisfy dependencies)
:: unable to satisfy dependency 'dbus-units' required by systemd
:: unable to satisfy dependency 'dbus-units' required by systemd
:: unable to satisfy dependency 'dbus-units' required by systemd

mdevaev, 2024-01-10:
This is upstreamer problem, please retry after few hours.

HB9VQQ, 2024-01-10:
Thank you it is working now

mdevaev, 2024-01-10:
:ok_hand:

closed 2024-01-10 by mdevaev