pikvm/pikvm #1211: Packages upgrade failed: libgpiod1 and libgpiod are in conflict

asad26, 2024-01-12:
**Describe the bug**
Getting an error while upgrading packages on PiKVM using `pacman -Syu`. It seems that PiKVM is trying to bring an older version of `libgpiod` package. See the error below:

```
[root@pikvm1 ~]# pacman -Syu
:: Synchronizing package databases...
 core                                                                                          245.4 KiB   982 KiB/s 00:00 [##########################################################################] 100%
 extra                                                                                           9.0 MiB  23.6 MiB/s 00:00 [##########################################################################] 100%
 community                                                                                      45.0   B   562   B/s 00:00 [##########################################################################] 100%
 alarm                                                                                          95.6 KiB  2.33 MiB/s 00:00 [##########################################################################] 100%
 aur                                                                                             9.3 KiB  51.8 KiB/s 00:00 [##########################################################################] 100%
 pikvm                                                                                          46.1 KiB   177 KiB/s 00:00 [##########################################################################] 100%
 tools                                                                                         770.0   B  3.96 KiB/s 00:00 [##########################################################################] 100%
:: Starting full system upgrade...
warning: python-ajsonrpc: local (1.2.0-4) is newer than extra (1.2.0-3)
resolving dependencies...
:: There are 2 providers available for dbus-units:
:: Repository core
   1) dbus-broker-units  2) dbus-daemon-units

Enter a number (default=1): 1
looking for conflicting packages...
error: unresolvable package conflicts detected
error: failed to prepare transaction (conflicting dependencies)
:: libgpiod1 and libgpiod are in conflict
```

Some info regarding `libgpiod`:

```
[root@pikvm1 ~]# pacman -Ss libgpiod
extra/libgpiod 2.0.1-1 [installed: 1.6.4-2]
    C library and tools for interacting with the linux GPIO character device
pikvm/libgpiod1 1.6.4-4
    C library and tools for interacting with the linux GPIO character device
```

**Expected behavior**
This `pacman -Syu` command should finish without errors and I would be able to upgrade all the packages.

**PiKVM info:**
  - Raspberry Pi board version: `RPi 4`
  - PiKVM platform: `v3-hdmi-rpi4`
  - KVMD version:
    ```
    kvmd 3.288-1
    kvmd-fan 0.26-1 
    kvmd-oled 0.26-1
    kvmd-platform-v3-hdmi-rpi4 3.288-1
    kvmd-webterm 0.47-1
    ```
  - uStreamer version: `5.46-1`
  - Linux kernel: `Linux pikvm1 6.1.61-1-rpi-ARCH #1 SMP Fri Nov  3 20:48:52 MSK 2023 armv7l GNU/Linux`

arch1mede, 2024-01-12:
Hi, Please consider joining our [Discord](https://discord.gg/bpmXfz5), we have a fix documented there already, most will be published there for future issues.

`pacman -S libgpiod` followed by `pacman -Syu`

closed 2024-01-12 by mdevaev

dropwhile, 2024-01-13:
I ran into this issue as well, and was able to find it via searching. I would not have found the solution if it was in discord (a service I have little interest in joining).
Thanks @asad26 (for filing the issue) and @arch1mede (for also providing a solution).

imp1sh, 2024-01-13:
For me it doesn't solve the issue to run `pacman -S libgpiod`. I don't want to join Discord as well.

mdevaev, 2024-01-14:
Please reflash the memory card and run `curl https://files.pikvm.org/update-os.sh | bash` to upgrade.

i-am-shodan, 2024-01-15:
If you've come here after googling the error messages above **please stop now**.

I ran the commands above and now my pikvm is broken and will need to be reimaged. Nginx is giving 500 errors.

asad26, 2024-01-15:
Thanks @arch1mede 
Running `pacman -S libgpiod` and then `pacman -Syu` solved the issue.

i-am-shodan, 2024-01-15:
not for me

paulctan, 2024-01-15:
Can confirm i-am-shodan 's report.  Doing `'pacman -S libgpiod` and then `pacman -Syu'` DOES NOT fix the problem.  

Trying the curl .. bash upgrade now.

i-am-shodan, 2024-01-15:
Nooooo don't do it. Your system won't come back up

paulctan, 2024-01-15:
Thanks for the warning i-am-shodan, but the system seems to have come back up fine after the curl .. bash upgrade.  Strange that you had that issue.

mdevaev, 2024-01-15:
@i-am-shodan if you encounter a problem, at least give me the logs so that I can solve it for you.

ThomasGerritsen, 2024-01-17:
I get the 500 error after this and when I run the script I get this. (via SSH)

`:: Synchronizing package databases...
 core                                  245.4 KiB   396 KiB/s 00:01 [#####################################] 100%
 extra                                   9.0 MiB  1801 KiB/s 00:05 [#####################################] 100%
 community                              45.0   B   500   B/s 00:00 [#####################################] 100%
 alarm                                  95.7 KiB   957 KiB/s 00:00 [#####################################] 100%
 aur                                     9.3 KiB  84.8 KiB/s 00:00 [#####################################] 100%
 pikvm                                  46.2 KiB  96.2 KiB/s 00:00 [#####################################] 100%
resolving dependencies...
looking for conflicting packages...
:: raspberrypi-utils and raspberrypi-firmware-pikvm are in conflict (raspberrypi-firmware). Remove raspberrypi-firmware-pikvm? [y/N] 
:: libgpiod1 and libgpiod are in conflict. Remove libgpiod? [y/N] 
:: kvmd and platformio are in conflict. Remove platformio? [y/N] 
error: failed to prepare transaction (could not satisfy dependencies)
:: installing python (3.11.6-1) breaks dependency 'python<3.11' required by python-ajsonrpc
:: installing python (3.11.6-1) breaks dependency 'python<3.11' required by python-luma-core
:: installing python (3.11.6-1) breaks dependency 'python<3.11' required by python-luma-oled
:: installing python (3.11.6-1) breaks dependency 'python<3.11' required by python-pyftdi
:: installing python (3.11.6-1) breaks dependency 'python<3.11' required by python-pyghmi
:: installing python (3.11.6-1) breaks dependency 'python<3.11' required by python-pyrad
:: installing python (3.11.6-1) breaks dependency 'python<3.11' required by python-pyserial-asyncio
:: installing python (3.11.6-1) breaks dependency 'python<3.11' required by python-raspberry-gpio
:: installing python (3.11.6-1) breaks dependency 'python<3.11' required by python-smbus2
:: installing python (3.11.6-1) breaks dependency 'python<3.11' required by python-spidev`

ThomasGerritsen, 2024-01-17:
Adding this: 

`root@pikvm ~]# pacman -Syu
:: Synchronizing package databases...
 core is up to date
 extra is up to date
 community is up to date
 alarm is up to date
 aur is up to date
 pikvm is up to date
:: Starting full system upgrade...
:: Replace raspberrypi-bootloader-x-pikvm with pikvm/raspberrypi-bootloader-pikvm? [Y/n] 
resolving dependencies...
:: There are 2 providers available for dbus-units:
:: Repository core
   1) dbus-broker-units  2) dbus-daemon-units

Enter a number (default=1): 
looking for conflicting packages...
error: unresolvable package conflicts detected
error: failed to prepare transaction (conflicting dependencies)
:: raspberrypi-utils and raspberrypi-firmware-pikvm are in conflict
[root@pikvm ~]# `

mdevaev, 2024-01-17:
@ThomasGerritsen please contact me via PM on discord.

benbaker76, 2024-01-17:
> Please reflash the memory card and run `curl https://files.pikvm.org/update-os.sh | bash` to upgrade.

The `pacman` method didn't work for me but running the script did.

alexisfrjp, 2024-01-20:
> Hi, Please consider joining our [Discord](https://discord.gg/bpmXfz5), we have a fix documented there already, most will be published there for future issues.
> 
> `pacman -S libgpiod` followed by `pacman -Syu`

That also broke my pikvm, nginx returns "502 Bad Gateway" even after a reboot.

mdevaev, 2024-01-20:
@alexisfrjp Can you SSH and attach the output of `journalctl -u kvmd | cat`?

mdevaev, 2024-01-20:
I appeal to everyone: if something is broken, **please attach a log `journalctl -u kvmd | cat` or (even better) get in touch with me in discord so that I investigate the incident**.

I'm ready to fix the problem if someone gets in touch and gives me access to their device.

snorma01, 2024-01-22:
@mdevaev When I run pacman -S libgpiod, it succeeds, and then running pacman -Syu it asks the following:

libgpiod1 and libgpiod are in conflict. Remove libgpiod?

What is the correct response to this? I suspect that saying yes might be what broke the install for others?

snorma01, 2024-01-22:
Never mind, I am getting the 500 internal server error even before removing libgpiod.

snorma01, 2024-01-22:
> @mdevaev When I run pacman -S libgpiod, it succeeds, and then running pacman -Syu it asks the following:
> 
> libgpiod1 and libgpiod are in conflict. Remove libgpiod?
> 
> What is the correct response to this? I suspect that saying yes might be what broke the install for others?

Update: I went ahead and said yes to this question and completed the pacman -Syu update and rebooted, and it fixed the 500 error.

asad26, 2024-01-23:
@snorma01 Yes, it was same for me and running `pacman -Syu` asks to remove `libgpiod` and I said `yes` in response. And I believe it is for removing some older incompatible version of `libgpiod`. 

Once the upgrade process completed, I have a new version of `libgpiod` installed. See:
```
[root@pikvm1 ~]# pacman -Ss libgpiod
extra/libgpiod 2.0.1-1
    C library and tools for interacting with the linux GPIO character device
pikvm/libgpiod1 2.1-1 [installed]
    C library and tools for interacting with the linux GPIO character device
```

mdevaev, 2024-01-23:
I've modified the script, should work fine now for old distros.

PiKVM has switched to using the 2.x API, so our alternative `libgpiod1` package exists temporarily, although it contains 2.1.

joshuataylor, 2024-01-23:
Sorry for the "me too", but the curl command fixed it, I have a "DIY PiKVM V2 Instructions", "Raspberry Pi 4 Model B"... but as usual please back anything you have up and have a recovery plan before running scary rando commands :).

yewkay, 2024-02-13:
I am here to post a #meetoo encountering the above issue. I used the curl command and it fixed the issue. Hope this helps others.