pikvm/pikvm #1212: Unifi POE GPIO Driver - Working POC - Gauging Interest

appkins, 2024-01-12:
**Is your feature request related to a problem? Please describe.**
I would like to control the POE state of a Unifi switch natively within pikvm. I have a working prototype for this, and am willing to clean up/change the code for a suitable PR.

**Describe the solution you'd like**
Ideally, all modes for Redfish power management would be supported. The Unifi API supports cycling and power states, so this is all demo'd in my POC.

**Describe alternatives you've considered**
At the moment, I'm overwriting the files in the python lib for kvmd. This is not ideal. I had originally considered a cmd based gpio implementation, but the flexibility of a native driver is working much better.

**Additional context**
More of a side note, but I also rewrote the Redfish implementation to support all configured power drivers. This will likely need much more guidance and review from the project owners, but very interested to discuss this side more in depth. Happy to open a separate issue.

My current work can be found in the following fork/branch. If you have time to look over it, please feel free to leave any feedback. I'm looking for general tips/guidance before I begin the PR process, as it's likely this could use a rework before I push anything for formal review.

[https://github.com/appkins-org/kvmd](https://github.com/appkins-org/kvmd/tree/cycle)

mdevaev, 2024-01-13:
Hi! I love Unifi devices, so having such a plugin would be great. I have briefly reviewed your changes and can suggest that you send me only the gpio plugin first, after which I will do some refactoring and we can continue further.

PS: I would prefer not to add hacks and various things specific to your environment. If you need to run and test something, then there is a pytest in the repository for this.

appkins, 2024-01-14:
> Hi! I love Unifi devices, so having such a plugin would be great. I have briefly reviewed your changes and can suggest that you send me only the gpio plugin first, after which I will do some [...]

Awesome, sounds great. I will split this out into a single purpose branch for the Unifi driver and create a PR.

mdevaev, 2024-01-14:
Please don't include API change in it. Only gpio.

appkins, 2024-01-22:
Will do, please disregard any mention of Redfish. This needs a dedicated design discussion the more I explore. For now, I'm only planning to add the gpio PR.

mdevaev, 2025-02-10:
@appkins Are you still interested?

mdevaev, 2025-10-28:
Closing due no feedback for a year.

closed 2025-10-28 by mdevaev