pikvm/pikvm #1213: Latest update 3.291 caused USB HDMI to stop working

chchia, 2024-01-14:
**Describe the bug**
USB capture card display for few second the become only black screen. as below gif file.

**To Reproduce**
updated to 3.291 today. was ok before this.

**PiKVM info:**
pacman -Syu all to latest today.

[image]

mdevaev, 2024-01-14:
Please attach the log. Click `System -> Log`.

chchia, 2024-01-14:
thanks for prompt response, this is system log

`[2024-01-14 04:00:52 kvmd.service] --- kvmd.apps.kvmd.auth               INFO --- Authorized user 'admin' via auth service 'htpasswd'
[2024-01-14 04:00:52 kvmd.service] --- kvmd.apps.kvmd.api.auth           INFO --- Logged in user 'admin'
[2024-01-14 04:00:52 kvmd.service] --- aiohttp.access                    INFO --- [- / 192.168.1.51] 'POST /auth/login HTTP/1.0' => 200; size=291 --- referer='https://192.168.1.3/login/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0'
[2024-01-14 04:00:52 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=194 --- referer='https://192.168.1.3/login/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0'
[2024-01-14 04:00:52 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=194 --- referer='https://192.168.1.3/login/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0'
[2024-01-14 04:00:52 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 192.168.1.51] 'GET /info?fields=auth,meta,extras HTTP/1.0' => 200; size=2160 --- referer='https://192.168.1.3/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0'
[2024-01-14 04:00:52 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=194 --- referer='https://192.168.1.3/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0'
[2024-01-14 04:00:53 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=194 --- referer='https://192.168.1.3/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0'
[2024-01-14 04:00:53 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=194 --- referer='https://192.168.1.3/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0'
[2024-01-14 04:00:53 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=194 --- referer='https://192.168.1.3/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0'
[2024-01-14 04:00:53 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 192.168.1.51] 'GET /auth/check HTTP/1.0' => 200; size=194 --- referer='https://192.168.1.3/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0'
[2024-01-14 04:00:53 kvmd.service] --- kvmd.apps.kvmd.server             INFO --- Registered new client session: WsSession(id=2997120624, {'stream': True}); clients now: 1
[2024-01-14 04:00:53 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Starting streamer ...
[2024-01-14 04:00:54 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- PRE-START-CMD: /bin/true pre-start
[2024-01-14 04:00:54 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Started streamer pid=710: /usr/bin/ustreamer --device=/dev/kvmd-video --persistent --format=mjpeg --resolution=1920x1080 --desired-fps=40 --drop-same-frames=30 --last-as-blank=0 --unix=/run/kvmd/ustreamer.sock --unix-rm --unix-mode=0660 --exit-on-parent-death --process-name-prefix=kvmd/streamer --notify-parent --no-log-colors --sink=kvmd::ustreamer::jpeg --sink-mode=0660
[2024-01-14 04:00:54 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [83.375      main] -- Starting PiKVM uStreamer 5.46 ...
[2024-01-14 04:00:54 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [83.375      main] -- Using internal blank placeholder
[2024-01-14 04:00:54 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [83.375      main] -- Using JPEG-sink: kvmd::ustreamer::jpeg
[2024-01-14 04:00:54 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [83.376      main] -- Listening HTTP on UNIX socket '/run/kvmd/ustreamer.sock'
[2024-01-14 04:00:54 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [83.376    stream] -- Using V4L2 device: /dev/kvmd-video
[2024-01-14 04:00:54 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [83.376    stream] -- Using desired FPS: 40
[2024-01-14 04:00:54 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [83.376      http] -- Starting HTTP eventloop ...
[2024-01-14 04:00:54 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => ================================================================================
[2024-01-14 04:00:54 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [83.610    stream] -- Device fd=9 opened
[... cut ...]
[2024-01-14 04:01:06 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=194 --- referer='https://192.168.1.3/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0'
[2024-01-14 04:01:06 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [95.916      http] -- HTTP: NEW client (now=1): [192.168.1.51]:0, id=57e2a0450474b9d5`

mdevaev, 2024-01-14:
Looks fine. Try incognito mode in browser, clear cache. If it doesn't help, check the javascript console in the browser for any errors.

chchia, 2024-01-14:
incognito mode have the same behavior, and this is javascript console output

```
bb.js:151 ===== Adding CSS: /share/css/x-desktop.css
tools.js:369 [04:07:29.564] LOG/INFO -- Keymap at div#keyboard-window: enabled Fix-Win-AltGr
tools.js:369 [04:07:29.948] LOG/INFO -- Stream: Janus WebRTC state: enabled=false, webrtc=true, h264=probably, imported=false
tools.js:369 [04:07:30.217] LOG/INFO -- Stream [MJPEG]: Using advance_headers=1 to fix Blink bugs
tools.js:369 [04:07:30.217] LOG/INFO -- Stream [MJPEG]: Refreshing ...
tools.js:369 [04:07:30.371] LOG/INFO -- Stream [MJPEG]: Found acceptable stream_client cookie: FMfz7CE0hnVdAeSq/4cd38028a98bba47
tools.js:369 [04:07:30.371] LOG/INFO -- Stream [MJPEG]: Active
tools.js:369 [04:07:45.472] LOG/INFO -- Terminal opened:  /extras/webterm/ttyd
extras/webterm/ttyd/token/:1 
        
        
        Failed to load resource: the server responded with a status of 404 ()
 [ttyd] websocket connection opened
 [ttyd] WebGL renderer loaded
 [ttyd] Leave site alert disabled
tools.js:369 [04:07:51.448] LOG/INFO -- Terminal closed
tools.js:369 [04:08:36.181] LOG/INFO -- Stream [MJPEG]: Inactive
tools.js:369 [04:08:36.287] LOG/INFO -- Stream [MJPEG]: Using advance_headers=1 to fix Blink bugs
tools.js:369 [04:08:36.287] LOG/INFO -- Stream [MJPEG]: Refreshing ...
streamer/stream?key=u4yIoKIh7EmJCwgW&advance_headers=1:1 
        
        
        Failed to load resource: the server responded with a status of 502 ()
tools.js:369 [04:08:38.618] LOG/INFO -- Stream [MJPEG]: Using advance_headers=1 to fix Blink bugs
tools.js:369 [04:08:38.618] LOG/INFO -- Stream [MJPEG]: Refreshing ...
tools.js:369 [04:08:39.524] LOG/INFO -- Stream [MJPEG]: Found acceptable stream_client cookie: c3mAEg5QMnRxNKEn/eea5acb365cde637
tools.js:369 [04:08:39.524] LOG/INFO -- Stream [MJPEG]: Active
tools.js:369 [04:08:48.717] LOG/INFO -- Terminal opened:  /extras/webterm/ttyd
extras/webterm/ttyd/token/:1 
        
        
        Failed to load resource: the server responded with a status of 404 ()
 [ttyd] websocket connection opened
 [ttyd] WebGL renderer loaded
 [ttyd] Leave site alert disabled
tools.js:369 [04:09:34.967] LOG/INFO -- Terminal closed
tools.js:369 [04:09:40.951] LOG/INFO -- Stream [MJPEG]: Inactive
tools.js:369 [04:09:41.059] LOG/INFO -- Stream [MJPEG]: Using advance_headers=1 to fix Blink bugs
tools.js:369 [04:09:41.059] LOG/INFO -- Stream [MJPEG]: Refreshing ...
stream:1 
        
        
        Failed to load resource: the server responded with a status of 502 ()
tools.js:369 [04:09:43.330] LOG/INFO -- Stream [MJPEG]: Using advance_headers=1 to fix Blink bugs
tools.js:369 [04:09:43.330] LOG/INFO -- Stream [MJPEG]: Refreshing ...
tools.js:369 [04:09:44.233] LOG/INFO -- Stream [MJPEG]: Found acceptable stream_client cookie: 7pcZoWFooD2F3pZY/40d8c773bdeb4cf2
tools.js:369 [04:09:44.233] LOG/INFO -- Stream [MJPEG]: Active
tools.js:369 [04:10:10.556] LOG/INFO -- Stream [MJPEG]: Inactive
tools.js:369 [04:10:10.659] LOG/INFO -- Stream [MJPEG]: Using advance_headers=1 to fix Blink bugs
tools.js:369 [04:10:10.659] LOG/INFO -- Stream [MJPEG]: Refreshing ...
stream:1 
        
        
        GET https://192.168.1.3/streamer/stream?key=KV4JpX66mpZUUdet&advance_headers=1 502 (Bad Gateway)
Image (async)
__checkStream @ stream_mjpeg.js:153
setInterval (async)
__ensureChecking @ stream_mjpeg.js:107
MjpegStreamer.self.ensureStream @ stream_mjpeg.js:65
__resetStream @ stream.js:257
(anonymous) @ stream.js:283
Promise.then (async)
__clickResetButton @ stream.js:281
el.onclick.el.ontouchend @ tools.js:118
tools.js:369 [04:10:12.879] LOG/INFO -- Stream [MJPEG]: Using advance_headers=1 to fix Blink bugs
tools.js:369 [04:10:12.879] LOG/INFO -- Stream [MJPEG]: Refreshing ...
tools.js:369 [04:10:13.781] LOG/INFO -- Stream [MJPEG]: Found acceptable stream_client cookie: S16T8wvfskAZMJ7L/300ffec9ace07ae5
[... 164 lines cut ...]
[2024-01-14 04:10:12 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [204.476      main] -- Starting PiKVM uStreamer 5.48 ...
[2024-01-14 04:10:12 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [204.476      main] -- Using internal blank placeholder
[2024-01-14 04:10:12 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [204.476      main] -- Using JPEG-sink: kvmd::ustreamer::jpeg
[2024-01-14 04:10:12 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [204.476      main] -- Listening HTTP on UNIX socket '/run/kvmd/ustreamer.sock'
[2024-01-14 04:10:12 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [204.476    stream] -- Using V4L2 device: /dev/kvmd-video
[2024-01-14 04:10:12 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [204.476    stream] -- Using desired FPS: 40
[2024-01-14 04:10:12 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [204.476      http] -- Starting HTTP eventloop ...
[2024-01-14 04:10:12 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => ================================================================================
[2024-01-14 04:10:12 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [204.477    stream] -- Device fd=9 opened
[2024-01-14 04:10:12 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [204.477    stream] -- Using capture type: single-planar
[... cut ...]
[2024-01-14 04:10:12 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=194 --- referer='https://192.168.1.3/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0'
[2024-01-14 04:10:12 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [204.704      http] -- HTTP: NEW client (now=1): [192.168.1.51]:0, id=300ffec9ace07ae5```

chchia, 2024-01-14:
below is latest system log with upgrade to 3.291 again using pacman -Syu

```[2024-01-14 04:07:20 kvmd.service] --- aiohttp.access                    INFO --- [- / -] 'GET /auth/check HTTP/1.0' => 401; size=285 --- referer='-'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0'
[2024-01-14 04:07:28 kvmd.service] --- kvmd.apps.kvmd.auth               INFO --- Authorized user 'admin' via auth service 'htpasswd'
[2024-01-14 04:07:28 kvmd.service] --- kvmd.apps.kvmd.api.auth           INFO --- Logged in user 'admin'
[2024-01-14 04:07:28 kvmd.service] --- aiohttp.access                    INFO --- [- / 192.168.1.51] 'POST /auth/login HTTP/1.0' => 200; size=318 --- referer='https://192.168.1.3/login/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0'
[2024-01-14 04:07:28 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=194 --- referer='https://192.168.1.3/login/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0'
[2024-01-14 04:07:28 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=194 --- referer='https://192.168.1.3/login/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0'
[2024-01-14 04:07:28 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 192.168.1.51] 'GET /info?fields=auth,meta,extras HTTP/1.0' => 200; size=2160 --- referer='https://192.168.1.3/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0'
[2024-01-14 04:07:28 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=194 --- referer='https://192.168.1.3/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0'
[2024-01-14 04:07:29 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=194 --- referer='https://192.168.1.3/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0'
[2024-01-14 04:07:29 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=194 --- referer='https://192.168.1.3/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0'
[2024-01-14 04:07:29 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=194 --- referer='https://192.168.1.3/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0'
[2024-01-14 04:07:29 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 192.168.1.51] 'GET /auth/check HTTP/1.0' => 200; size=194 --- referer='https://192.168.1.3/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0'
[2024-01-14 04:07:29 kvmd.service] --- kvmd.apps.kvmd.server             INFO --- Registered new client session: WsSession(id=3010158192, {'stream': True}); clients now: 1
[2024-01-14 04:07:29 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Starting streamer ...
[2024-01-14 04:07:29 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- PRE-START-CMD: /bin/true pre-start
[2024-01-14 04:07:29 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Started streamer pid=651: /usr/bin/ustreamer --device=/dev/kvmd-video --persistent --format=mjpeg --resolution=1920x1080 --desired-fps=40 --drop-same-frames=30 --last-as-blank=0 --unix=/run/kvmd/ustreamer.sock --unix-rm --unix-mode=0660 --exit-on-parent-death --process-name-prefix=kvmd/streamer --notify-parent --no-log-colors --sink=kvmd::ustreamer::jpeg --sink-mode=0660
[2024-01-14 04:07:29 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [41.717      main] -- Starting PiKVM uStreamer 5.48 ...
[2024-01-14 04:07:29 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [41.717      main] -- Using internal blank placeholder
[2024-01-14 04:07:29 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [41.717      main] -- Using JPEG-sink: kvmd::ustreamer::jpeg
[2024-01-14 04:07:29 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [41.718      main] -- Listening HTTP on UNIX socket '/run/kvmd/ustreamer.sock'
[2024-01-14 04:07:29 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [41.718    stream] -- Using V4L2 device: /dev/kvmd-video
[2024-01-14 04:07:29 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [41.718    stream] -- Using desired FPS: 40
[2024-01-14 04:07:29 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [41.718      http] -- Starting HTTP eventloop ...
[... cut ...]
[2024-01-14 04:10:12 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=194 --- referer='https://192.168.1.3/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0'
[2024-01-14 04:10:12 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [204.704      http] -- HTTP: NEW client (now=1): [192.168.1.51]:0, id=300ffec9ace07ae5```

chchia, 2024-01-14:
[image]

chchia, 2024-01-14:
just tried using mobile phone browser, same behavior, suspect is not client side problem.

chchia, 2024-01-14:
@mdevaev I am pulling all my hair out.....

i just power reboot the PIKVM by total power off and on again, and it is now working....

previously a "reboot" command did not resolve the problem, and total power recycling is required... strange!

mdevaev, 2024-01-14:
Very weird, I recimmend to replace the memory card. Usually, all the weirdest errors are related to data corruption.

closed 2024-01-14 by mdevaev