pikvm/pikvm #1216: Upgrade decision between dbus-broker-units and dbus-daemon-units

rgov, 2024-01-17:
I obtained a PiKVM V4 Mini recently and decided to do a software update out of the box, following the [First Steps](https://docs.pikvm.org/first_steps/) documentation:

```
$ rw
$ pacman -Syu
...
:: There are 2 providers available for dbus-units:
:: Repository core
   1) dbus-broker-units  2) dbus-daemon-units
```

I went to a search engine to find the right selection, but didn't find an explicit answer. **I believe the correct choice is dbus-broker-units.**

A search engine brought me to #1211, related to libgpiod, but that seems to be a red herring. Even after installing libgpiod, the user is still presented with the same choice.

Therefore, according to the advice in the FAQ, I ran the `update-os.sh` script. I observed that it installs the `dbus-broker-units` package. Therefore, I probably should have selected that to begin with.

mdevaev, 2024-01-17:
Hello. You should choose (1). Defaults are always a good choice.

closed 2024-01-17 by mdevaev

rgov, 2024-01-18:
(1) is not the default, it is just the first choice.