pikvm/pikvm #1222: Quick win/easy to implement: Update command like pihole "pihole -up"

Majestic7979, 2024-01-23:
Maxim, thanks for your contiuned development and support of the PiKVM, I honestly can't put in words how much I appreciate you for creating and maitaining this.

You probably know about the PiHole, right? To update it, you just SSH into the device running it, the the command below and press enter:
```
pihole -up
```
And then a script is invoked somewhere, and it spits out the status on the screen. Simple and job done.

I noticed that doing pacman -Syu on the PiKVM lately has been throwing errors, and that it's necessary to do a curl command to start a script from your site.

Since the script seems to be hosted always on the same path, could you please give us a similar thing in the PiKVM?

Maybe the command could imitatee the PiHole and be
```
pikvm -update
```

By typing that and hitting enter, the PiKVM would curl and execute that script (provided the user has entered root and put the file system in rw mode, of course).

I don't imagine this would take more than 20 minutes of your time and would provide a much easier, better way of updating properly than using pacman and potentially breaking things.

Please give it a thought and if you can make it happen on your next update release that would be super cool!

Thanks again.

mdevaev, 2024-01-26:
Hello! In fact, I've already thought about it, and I'm going to implement it in the coming days

Majestic7979, 2024-02-06:
Hi @mdevaev not rushing you, please take your time. Just wanna know if you managed to release this? It's because I updated the system using the curl command and was wondering if the update included this change? If so what command do you intend to be using for this feature? I'll try it on future updates :) Thanks!

mdevaev, 2024-02-06:
I need to take a careful approach to the release of this so as not to break anything. The script should handle various situations and correctly perform rootfs remounting if necessary.

Majestic7979, 2024-02-07:
> I need to take a careful approach to the release of this so as not to break anything. The script should handle various situations and correctly perform rootfs remounting if necessary.

That's cool Maxim, no rush, I was just wondering what is the actual command you will choose to use with this, will it be "pikvm -update" or something else?

mdevaev, 2024-02-07:
Probably `pikvm-update`, one word.

mdevaev, 2024-02-17:
It's done. Install the update with this:
```
curl https://files.pikvm.org/update-os.sh | bash
```

After that, you will receive the command `pikvm-update` which can be used for all future updates.

closed 2024-02-17 by mdevaev

Majestic7979, 2024-02-19:
Done and works like a charm, thanks very much for always listening to your user's suggestions and needs. I hope you manage to fix that issue with black screens/disconnections in Chrome WebRTC for us soon 🙏

mdevaev, 2024-02-19:
:ok_hand: