pikvm/pikvm #1228: OCR not working when using http

ich777, 2024-01-26:
**Describe the bug**
I've configured PiKVM for http only and OCR is missing:

**To Reproduce**
I disabled https like described [here](https://docs.pikvm.org/faq/#usb-problems-keyboard-mouse-mass-storage-etc) at the section `I can't use the PiKVM web interface on iOS: the Web UI network indicator flashes yellow` <- Yes I know the security implications but I'm only using it on my local network or through a VPN.

**Expected behavior**
OCR working with http only

**Screenshots**
[image]

**Desktop:**
 - OS: Debian Bullseye/Windows 11
 - Browser Firefox/Edge/Chrome
 - Version: latest version for all three listed above as time of writing
 - VNC client (if used) <- not used

**PiKVM info:**
  - Raspberry Pi board version RPi 4
  - PiKVM platform v2-hdmi
  - Video capture type USB adapter
  - KVMD version: 3.288
  - uStreamer version: 4.9

**Additional context**
Is this a technical limitation or just a bug? Fell free to close that issue if it's a bug.

mdevaev, 2024-01-27:
Hello. It should be fixed now. Please update and test:
```
curl https://files.pikvm.org/update-os.sh | bash
```

ich777, 2024-01-27:
Thank you! Working perfectly fine!

mdevaev, 2024-01-27:
:ok:

ich777, 2024-01-27:
Sorry, I have to reopen the issue... I have not actually tried OCR just checked if I can see it... 

I now see the OCR feature through http but it actually doesn't work, you'll see in this gif that the gear never stops turning:
[image]

When I connect through https it is working just fine:
[image]

mdevaev, 2024-01-27:
Please open developer tools and take a screenshot of JS console and requests list

ich777, 2024-01-27:
I think this is what you where after:
[image]

mdevaev, 2024-01-27:
Try again with 3.296

ich777, 2024-01-27:
Thank you, now it's working!

closed 2024-01-27 by ich777