pikvm/pikvm #1230: [QUestion] Why does PiKVM generate high bit-rate on a static screen?

yurivict, 2024-01-29:
**Describe the bug**

With an almost completely static screen the bit-rate alternates between hundreds of kbps:
[image]
and several Mbps:
[image]

```iftop``` shows several Mbps fro  the PiKVM host.

When the screen is static - why does PiKVM generate such a high bit-rate?

The screen is a KDE desktop with only a clock running and changing a small area once a minute.

**Expected behavior**
Bit-rate should increase when screen changes a lot, and should be very low when there are no screen changes.

**Desktop (please complete the following information):**
 - OS: FreeBSD 14.0
 - Browser: chrome

**PiKVM info:**
  - Raspberry Pi board version RPi 4
  - PiKVM platform v2-hdmi DIY
  - Video capture type: CSI bridge
  - KVMD version: ```
  kvmd 3.284-1
  kvmd-fan 0.26-1
  kvmd-oled 0.26-1
  kvmd-platform-v2-hdmi-rpi4 3.284-1
  kvmd-webterm 0.44-1
```
  - uStreamer version: ```5.45-1```
  - Linux kernel: ```Linux pikvm 6.1.61-1-rpi-ARCH #1 SMP Fri Nov  3 20:48:52 MSK 2023 armv7l GNU/Linux```

mdevaev, 2024-01-29:
I suppose the reason is keyframe interval.By default, we send it every second. You can reduce this and see what happens. In the `System` menu, set the `H.264 gop` parameter to 0.

yurivict, 2024-01-29:
With the ```H.264 gop``` parameter being 0 the bit-rate went down to between 170 kbps and 255 kbps for a static screen.

But is this still too high? For each new frame it should essentially say "nothing new" most of the time. So what is transmitted in these hundred+ kbps?

mdevaev, 2024-01-29:
I don't know. We use Raspberry hardware encoder (to reduce latency), it's a black box.

closed 2024-01-29 by yurivict