pikvm/pikvm #1233: [question] Is there a way to disable https and to switch to http?

yurivict, 2024-01-31:
**Describe the bug**
A clear and concise description of what the bug is.
I run PiKVM w/in LAN between hosts that aren't exposed through the router, so https isn't really needed.
Performance might improve if encryption/decryption wouldn't occur.
Is there an easy way to disable https, and to switch to http?

The [first steps](https://docs.pikvm.org/first_steps/) guide doesn't talk about this.

mdevaev, 2024-02-01:
In my experience, this does not provide any performance benefits.

You can edit `/etc/kvmd/nginx.conf` if you want get rid of https.

mdevaev, 2024-02-03:
Since 3.301 you can simply disable HTTPS using the override.yaml: https://www.reddit.com/r/pikvm/comments/1ahxvhh/kvmd_3301_generated_nginx_configs/

closed 2024-02-03 by mdevaev