pikvm/pikvm #1238: PiKVM 4 Plus hardware case issue with official RPi CM4 antenna kit

wb6vpm, 2024-02-02:
**Describe the bug**
Unable to install the official CM4 antenna kit, as the hole for the antenna is too small (yes, I did make sure that I was attempting to do it with the flat spot oriented correctly).

**To Reproduce**
Steps to reproduce the behavior, like:

1. Attempt to install SMA antenna connector into either one of the 2 holes on the case.
2. SMA connector will not go through the hole, as the hole is too small

**Expected behavior**
SMA connector goes through the hole

**Screenshots**
(multiple pictures)
https://share.icloud.com/photos/0d2-hayiwKevMLQ2CED75iJCg

**Desktop (please complete the following information):**
NA, hardware issue

**PiKVM info:**
Brand new PiKVM 4 Plus kit

**Additional context**
Both items (PiKVM 4+ & CM4 antenna kit) were received this week from pishop.us webstore

mdevaev, 2024-02-03:
Hello. Did you try just to screw it with some force? Most likely the painting is too thick.

wb6vpm, 2024-02-03:
> Hello. Did you try just to screw it with some force? Most likely the painting is too thick.

It’s a keyed hole (has flat side on it, making it kind of look like a D, instead of an O), which is done specifically to prevent that from happening, so unfortunately, there is no way to screw it in.

mdevaev, 2024-02-03:
What about pushing it through? I still think the problem is the paint.

wb6vpm, 2024-02-03:
I gave it a pretty hard push against it, but I can try again on Monday. 

Out of curiosity, according to the design files for the case, what size hole is it supposed to be? I ask because it’s almost like they inadvertently grabbed the wrong dimension off the connector (such as the tip instead of the threaded part) as their basis for the hole.

mdevaev, 2024-02-03:
The hole was correct in the drawings and in the first batches, we checked it. If there is a problem now, it is either painting, or the factory messed up with a new batch and we will deal with it.

wb6vpm, 2024-02-03:
Sounds good. After I attempt the push through, I’ll go ahead and scrape the paint off inside the hole to see if that fixes it and then let you know.

wb6vpm, 2024-02-08:
Sorry about that, this week has just gotten away from me. I did test it, and with scraping the paint off, it does fit, barely, with a pretty significant amount of force needed (comparatively, given the small size and delicacy of the part). I would recommend on the next run, going up 0.5mm to allow for the paint and tightness of the hole in general. Also as an aside, I would recommend posting the recommended cable routing path in the documentation, or alternatively to all of this, just include the antenna kit pre-installed in the next revision, and add $10 to the cost (or whatever it takes to offset the additional parts/labor costs), because honestly, getting to that connector, and then getting it seated on the CM4 board is a complete PITA once the PiKVM is assembled.

mdevaev, 2024-02-08:
Thank you for the information and I'm sorry for this inconvenience.

I will pass this on to the developer of the case and we will fix it in the next batch.

closed 2024-02-08 by mdevaev