pikvm/pikvm #1242: HP Z2 doesn't show Boot Menu

andybowser, 2024-02-12:
**Describe the bug**
PiKVM connected via HDMI to Mini Displayport cable to HP. 
On boot I push F9 to go to Boot Menu, which works, but then I can't see the boot menu. It shows the black HP screen. Taking HDMI out of PiKVM into a monitor, shows the menu fine.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Reboot or turn on PC
2. Press F9
3. No Boot Menu

**Expected behavior**
Showing the boot menu as per this screenshot.
[image]

**Screenshots**
I just see the following
[image]

**Desktop (please complete the following information):**
 - OS: macOS
 - Browser Firefox
 - Version 122.0

**PiKVM info:**
  - Raspberry Pi board version RPi 4
  - kvmd 3.173-1
kvmd-fan 0.22-1
kvmd-oled 0.13-1
kvmd-platform-v3-hdmi-rpi4 3.173-1
kvmd-webterm 0.42-1

mdevaev, 2024-02-12:
Which pikvm device do you use?

andybowser, 2024-02-12:
Pikvm v3

mdevaev, 2024-02-13:
Please try some alternative EDIDs: https://docs.pikvm.org/edid

andybowser, 2024-02-16:
Thank you. The Intel NUC EDID worked.

closed 2024-02-16 by andybowser

mdevaev, 2024-02-16:
@andybowser Could you also check the new official EDID? https://www.reddit.com/r/pikvm/comments/1ars4lx/kvmd_3304_video_quality_and_compatibility/

andybowser, 2024-04-25:
> @andybowser Could you also check the new official EDID? https://www.reddit.com/r/pikvm/comments/1ars4lx/kvmd_3304_video_quality_and_compatibility/

Yes. The new EDID works. 
I also bought the V4 for a project and that works fine with default config.
Thanks

mdevaev, 2024-04-26:
Great!