pikvm/pikvm #1244: USB-passthrough - pikvm to server

4plb0mb, 2024-02-12:
**Is your feature request related to a problem? Please describe.**
I need a way to get data (mostly backup files) from my client to the server (server in this case being the machine the pikvm is connected to) and back. I'm aware of #672, but afaik it's not _quite_ the same.

**Describe the solution you'd like**
My preferred solution would be a USB HDD connected to the pikvm which I can access via ssh to the pikvm (tailscale, whatever) AND which I can connect to the server. Similar solutions are fine, too, as long as both the server and client can be given access to the data (not at the same time). I mainly want that HDD so that the microSD doesn't get annihilated from all the writing I want to do.
[image]

**Describe alternatives you've considered**
I have tried .img or .vhd files which work to a certain degree - in a test, I mounted a folder from my Synology via fstab nfs on the pikvm, where I then stored a vhd. I was able to both access and write data to the vhd from my client and I could connect the .vhd to the server.
However, while the OS of the server can read the vhd if it's connected, Acronis True Image (the backup software I use for that server) can't and needs to initialize the vhd, rendering it useless for both the server and client OS.
[image]

**Additional context**
I have a few other ideas, but this is not exactly my area of expertise, so this is as close to a solution as I was able to get thus far. I considered having the pikvm somehow make the .vhd's file system available via network (nfs) so that you can directly read from/write to the .vhd while at the same time connecting it to the server, but I haven't made a lot of progress there, yet. Like I said, not exactly my area of expertise.

As a closing note, I want to add that this is my first post _ever_ on github. I've searched the existing issues for anything I could find regarding my issue and I've done some general research to the best of my abilities.
I sincerely hope that I haven't missed anything that would lead to me wasting anybody's time. Many thanks in advance!

mdevaev, 2024-03-14:
Hi. I understand what you want, but at the moment I have no idea how to do it with a little blood :/ I will keep it for the future.

4plb0mb, 2024-03-16:
> Hi. I understand what you want, but at the moment I have no idea how to do it with a little blood :/ I will keep it for the future.

That's absolutely fine! I'm insanely grateful for all the work you're putting into the project, if you get around to it, great, if you don't, still great. :)

steve-todorov, 2024-12-26:
I'm also interested in a USB pass-through functionality. I'd be open to help out with testing any potential solutions. :)