pikvm/pikvm #1245: Services *@etc-pacman.d-gnupg.socket are failing to start.

tobiashochguertel, 2024-02-12:
**Describe the bug**

After upgrading piKVM, some services didn't start correctly.

[image]

```
* dirmngr@etc-pacman.d-gnupg.socket                                                                                                          loaded failed failed    GnuPG network certificate manag>
  dm-event.socket                                                                                                                            loaded active listening Device-mapper event daemon FIFOs
* gpg-agent-browser@etc-pacman.d-gnupg.socket                                                                                                loaded failed failed    GnuPG cryptographic agent and p>
* gpg-agent-extra@etc-pacman.d-gnupg.socket                                                                                                  loaded failed failed    GnuPG cryptographic agent and p>
* gpg-agent-ssh@etc-pacman.d-gnupg.socket                                                                                                    loaded failed failed    GnuPG cryptographic agent (ssh->
* gpg-agent@etc-pacman.d-gnupg.socket                                                                                                        loaded failed failed    GnuPG cryptographic agent and p>
* keyboxd@etc-pacman.d-gnupg.socket                                                                                                          loaded failed failed    GnuPG public key management ser>
*
```

```
[root@pikvm ~]# systemctl status gpg-agent-browser@etc-pacman.d-gnupg.socket
x gpg-agent-browser@etc-pacman.d-gnupg.socket - GnuPG cryptographic agent and passphrase cache (access for web browsers) for /etc/pacman.d/gnupg
     Loaded: loaded (/usr/lib/systemd/system/gpg-agent-browser@.socket; static)
     Active: failed (Result: resources)
   Triggers: * gpg-agent@etc-pacman.d-gnupg.service
       Docs: man:gpg-agent(1)
     Listen: /etc/pacman.d/gnupg/S.gpg-agent.browser (Stream)

Jan 26 01:01:15 pikvm systemd[1]: gpg-agent-browser@etc-pacman.d-gnupg.socket: Failed to create listening socket (/etc/pacman.d/gnupg/S.gpg-agent.browser): Address already in use
Jan 26 01:01:15 pikvm systemd[1]: gpg-agent-browser@etc-pacman.d-gnupg.socket: Failed to listen on sockets: Address already in use
Jan 26 01:01:15 pikvm systemd[1]: gpg-agent-browser@etc-pacman.d-gnupg.socket: Failed with result 'resources'.
Jan 26 01:01:15 pikvm systemd[1]: Failed to listen on GnuPG cryptographic agent and passphrase cache (access for web browsers) for /etc/pacman.d/gnupg.
```

**To Reproduce**

1. Fresh installation of pikvm DIY v2.
2. `curl https://files.pikvm.org/update-os.sh | bash`
3. reboot

**Expected behavior**

No Services should fail when starting pikvm.

**PiKVM info:**
  - Raspberry Pi board version: `4 with 4 GB RAM`
  - PiKVM platform: `v2` `USB-HDMI Dongle`
  - Video capture type: `USB-HDMI Dongle`
  - KVMD version: `pacman -Q | grep kvmd`:
```
[root@pikvm ~]# pacman -Q | grep kvmd
kvmd 3.302-1
kvmd-fan 0.30-1
kvmd-oled 0.26-1
kvmd-platform-v2-hdmiusb-rpi4 3.302-1
kvmd-webterm 0.48-1
```
  - uStreamer version: `pacman -Q | grep ustreamer`:
```
[root@pikvm ~]# pacman -Q | grep ustreamer
ustreamer 5.49-1
```

  - Linux kernel: `uname -a`:
```
[root@pikvm ~]# uname -a
Linux pikvm 6.1.61-1-rpi-ARCH #1 SMP Fri Nov  3 20:48:52 MSK 2023 armv7l GNU/Linux
```

mdevaev, 2024-02-12:
It seems that there is nothing serious about this. Unfortunately, some components just have to be in rw mode to work properly.

mdevaev, 2024-03-17:
I've masked these services in OS.

closed 2024-03-17 by mdevaev