pikvm/pikvm #1246: Keyboard does not work when trying to enter BIOS

wolfdael, 2024-02-14:
**Describe the bug**
v2 DIY with a Pi 4b, latest pikvm software as of 2/14/24. Connecting via Chrome (latest) on Windows 10.

I was trying to enter the BIOS by spamming the DEL key in the web interface while the computer rebooted. It would not enter the BIOS and after the monitored computer booted up into the OS (unraid) the keyboard directly connected (USB) to this computer would not work. Rebooting the PiKVM made the local keyboard work again.

**To Reproduce**
Steps to reproduce the behavior, like:
Steps to reproduce:

Login to the PiKVM web interface and open KVM mode
Reboot the monitored computer
Start hitting the DEL key repeatedly as soon as the screen goes black
See the POST screen which says "Hit DEL to enter the BIOS"
BIOS is not entered
See the Unraid fully boot up
Reboot the PiKVM and see that the local keyboard now works again on the boot screen

I have tried all this after disabling MSD and issue still persists. I use my machine with unraid, so I cant say how pikvm works in other scenarios.

MSI Z90M MAG Z690M MORTAR WIFI -- BIOS -- 7D42vBH

arch1mede, 2024-02-14:
Hi, I see you are already on Discord discussing others issues. Lets continue this there, sounds like you have a few issues you have discussed.

closed 2024-02-16 by mdevaev

Eforen, 2024-11-15:
What was the fix?!

arch1mede, 2024-11-15:
> What was the fix?!

Hi, if you are having issues, please join [Discord](https://discord.gg/bpmXfz5to) to discuss, this Github shouldn't be used for support.