pikvm/pikvm #1247: Be able to adjust the mouse jiggler parameters...

Majestic7979, 2024-02-14:
Maxim I just noticed you managed to implement the jiggler, can I ask if this works even when the browser is closed? And is there a way to personalize it?

I currently have to use a macro... the macro presses a couple of keys even few minutes, as well as moves the mouse. This is necessary because the keys control a webpage which to remain live needs keyboard input as well as mouse movement.

So I assume if the Jiggler function from the system menu is based on the Macro engine this could be possible?

mdevaev, 2024-02-14:
> can I ask if this works even when the browser is closed? 

Yes, it is.

> And is there a way to personalize it?

Unfortunately no, this is a pretty deterministic function, and I don't think it will be done. Maybe in the future when we will support server-side macros, but jiggler should remain simple and oaky.

Majestic7979, 2024-02-15:
Thanks for confirming Maxim :) Hopefully we get something like that in the future then.

mdevaev, 2024-02-15:
:ok_hand:

closed 2024-02-15 by mdevaev