pikvm/pikvm #1248: Can't delete stuff even after making filesystem rw

Majestic7979, 2024-02-14:
```
[root@pikvm kvmd]# rm -Rf msd
rm: cannot remove 'msd/systemrescue-10.00-amd64.iso': Read-only file system
rm: cannot remove 'msd/debian-live-11.7.0-amd64-kde+nonfree.iso': Read-only file system
rm: cannot remove 'msd/debian-live-12.5.0-amd64-xfce.iso': Read-only file system
rm: cannot remove 'msd/lost+found': Read-only file system
[root@pikvm kvmd]# rw
[root@pikvm kvmd]# rm -Rf msd
rm: cannot remove 'msd/systemrescue-10.00-amd64.iso': Read-only file system
rm: cannot remove 'msd/debian-live-11.7.0-amd64-kde+nonfree.iso': Read-only file system
rm: cannot remove 'msd/debian-live-12.5.0-amd64-xfce.iso': Read-only file system
rm: cannot remove 'msd/lost+found': Read-only file system
```

Why am I doing this?
Because there's an image that refuses to be deleted from the WebUI even though it's clearly in the msd folder... when I press delete this is what happens:
[image]

So I navigated to /var/lib/kvmd/msd and tried to rw then rm <filename> but it still says it's read only, so I can't free up space because the image is undeletable... 

Not sure if this is a bug, but would be grateful for a quick and dirty fix until you fix it, because I really need to free up space, thank you!

mdevaev, 2024-02-14:
`/var/lib/kvmd/msd` is a special directory that controlled by kvmd. Never delete this directory, or kvmd will stop working. Btw It's on a separate partition, so I guess that's not your problem with running out of space. Better try to remove old packages cache: `rm -rf /var/cache/pacman/pkg`

Majestic7979, 2024-02-14:
Never mind Maxim, sorry for wasting your time.

I wanted to delete just the image. The above was just a test to show terminal output.

I found this, followed and fixed the issue - I was able to delete the image. I thought the "rw" applied to the whole file system...

https://docs.pikvm.org/msd/#manual-images-uploading

mdevaev, 2024-02-14:
:ok_hand: