pikvm/pikvm #1249: The paste "speed" is too fast, so characters are not being pasted as shown on the paste window. It's affecting changing passwords with passwd before installing openssh

Majestic7979, 2024-02-15:
Due to business requirements I'm having to change some systems passwords to a very large string...

So to test this, I used Bitwarden to generate a password with 64 words (using the word function, instead of character). This means I  get a password around 500 characters long (remember this is a test).

I paste the password into the Text > paste box in the PiKVM, then I make sure the layout is en-gb and hit paste... I can see that the password that gets pasted is not the same as was originally provided.

Why is this a big issue? Because on the terminal you can't see what you're typing... so you paste, change the password using passwd command, but next time it won't logon because the paste from the PiKVM won't be the same, there's always a small variation.

Here's an example I just generated, original text:

```
Sacrament-Excavate-Relish-Antirust-Monday-Eggplant-Spoilage-Shifting-Underpass-Another6-Squeeze-State-Pyromania-Eardrum-Curing-Professor-Directory-Elevating-Defective-Sneak-Dexterous-Zookeeper-Deskwork-Confiding-Drift-Dazzling-Durable-Geranium-Barstool-Recovery-Slapping-Equipment-Backslid-Pointless-Heave-Wince-Jot-Parlor-Greedy-Boss-Status-Plop-Idealize-Superman-Harvest-Spinal-Poster-Composite-Sighing-Boney-Scion-Volatile-Unclad-Shifter-Mushily-Cloak-Pushpin-Endeared-Capitol-Geranium
```

And here is what the PiKVM pastes into the application:

[image]

Can see right from the first character it's already wrong, with lowercase "sacrament" instead of "Sacrament". Obviously passwords will be case sensitive... 

It would be amazing if this could be looking into because it can have terrible ramifications - as in losing access to remote system because the password seemed to change correctly but since it was invisible it was impossible to confirm, so only way to confirm is logging off the user, but when logging back on the password isn't accepted... for something of that length it's impossible to even guess, the whole point of a difficult password ;-) but of course the user needs to know it haha.

Majestic7979, 2024-02-15:
PS. Just tested a couple of times and it seems it's mostly affecting the first character, so the first character is always lowercase... If I use Apple it will do apple, if I use Elevator it will do elevator, if I use Supercalifragilisticexpialidocious it will do supercalifragilisticexpialidocious...

mdevaev, 2024-02-15:
It seems this is the same problem: https://github.com/pikvm/pikvm/issues/858

Majestic7979, 2024-02-16:
I didn't test on any mobile device, only on Linux servers connected to the PiKVM directly (no multiport KVM in-between). Don't know what's causing the issue but definitely need addressing urgently if you're able, pretty please. This can create massive headache when administering remote locations and the password isn't accepted because of errors. But don't stress Maxim, you have a life to live too. I think you're amazing for taking such good care of this project :-) Thanks so much!

mdevaev, 2024-03-16:
I've come back to this problem again and I can't find any way to reproduce it. Is it only with linux hosts?

Majestic7979, 2024-03-17:
> Refer

I'm sysadmin for only one windows server, the rest are all debian linux, and it happened on debian. The root password I was setting during setup was massive as you can see below. Not sure if this is a buffer issue, is there any error correction when sending keypresses over the virtual usb keyboard to the host? I also have a problem where if I press Shift key to capitalize a work, sometimes the virtual keyboard from the UI will show that the Shift button remained pressed down even though I am physically not pressing it anymore on the physical keyboard. Seems like PiKVM is not polling the real keyboard properly if it's thinking a key is depressed when it's not.

mdevaev, 2024-03-17:
Did you try other browser?

Majestic7979, 2024-03-17:
For the shift thing both chrome and firefox. For the password thing just chrome.

Also something to add, on the shift button thing it normally happens when I am typing an acronym like AT&T

mdevaev, 2024-12-01:
I've made another go at this problem and still can't reproduce it :/

Has anything changed on your side since then? Is there still a problem?

Majestic7979, 2024-12-03:
It is an issue, it seems related to network speed, so if the KVM is connected over WiFi it is worse. If it's ethernet it is less worse. I was not having this issue for a while, but every week I run the update command to make sure I have the latest fixes, and it seems like even just normal typing is slower now. I am finding that the KVM is not passing all the keystrokes to the remote PC, so if I tried to type this sentence: "the quick brown fox jumped over the lazy dog" the remote PC would show this on the screen "te quik brn fox jmed over th lazy dg" or something similar, maybe not so many missed characters. But it should have some sort of buffer and it doesn't seem to. Missed characters just never get sent. I'd prefer to have a buffered output so that even if there is a delay on keyboard output at least the remote server can still receive all the data. Also, if there was some sort of indicator that the buffer is being sent, it could even be like the ATX icons that flash different colors, maybe a red dot that represents keyboard buffer being sent, which turns into a green dot when all the characters were sent to the remote device, that would help! Imagine pasting a huge block of text, and not knowing when the remote device finished receiving!? So having a visual indicator on the top bar would help, when the indicator dot is red = KVM still sending keyboard strokes to remote device... dot turned green = buffer is empty and no longer sending keystrokes... would be amazing! If you can and want to do it, please do I'd be soooo happy <3

mdevaev, 2024-12-03:
The difficulty is that the insert function actually uses a buffer (and TCP, of course) and sends it to PiKVM in its entirety. If what you see is happening, then for some reason the host keyboard driver ignores typing keys too fast.

Majestic7979, 2024-12-04:
> The difficulty is that the insert function actually uses a buffer (and TCP, of course) and sends it to PiKVM in its entirety. If what you see is happening, then for some reason the host keyboard driver ignores typing keys too fast.

Mind blown here. If that is the case then is it possible to adjust the config somehow to throttle the speed on the KVM side please? I mean I can understand how some hosts would not like thousands of characters being typed really quickly... raspberry pis for example, the older models could struggle to receive a ton of text quickly... I mean, even the fastest human being cannot type 500 characters per second, right? :-) I am exaggerating to be funny but you know what I mean :-) Would be nice to have more control over what the KVM does, so that it can be adapted for different situations.

But the rest of my message about being able to control what each component does (whether it uses TCP or UDP) would be amazing.

I still unfortunately see that dreaded Janus issue with the WebRTC stream cutting out when there's audio and the connection speed is slow (generally when the KVM is using WiFi rather than ethernet)... had to buy a switch only to accomodate the KVM in certain situations, but not all people I support will allow me to use their wired LAN because they don't want to change firewall rules bla bla bla 🙄 hard to argue with a client :(

Majestic7979, 2024-12-12:
> Added slow typing mode. It will be available in the next release (in hour). This is a simple but not the best solution with the addition of a small delay, and it does not guarantee getting rid of the problems completely, because I would have to completely redo all HID plugins to implement a queue with trottling.

Thanks Maxim, you know I always appreciate your work highly, can I ask if despite the closure you are planning at some point to rewrite the stuff, if that would make the fix more reliable? I want to clarify that I am not putting any pressure on you, just wanna be aware if it is being considered at all or not.

mdevaev, 2024-12-12:
Not unless it is critically needed to fix the problem.

Majestic7979, 2024-12-12:
> Not unless it is critically needed to fix the problem.

Got it. I would never pretend something is urgent unless it is, so look forward to testing next week and hopefully it remains all that is needed and no further work for you, you are already far busy as it is :)