pikvm/pikvm #1252: toggle reset to cmos reset button label in gui

realies, 2024-02-17:
found this pretty useful when having fun with overclocking, guess it could be useful for other applications
[image]

mdevaev, 2024-02-18:
You can use [gpio menu functions](https://docs.pikvm.org/gpio/) for this.

closed 2024-02-19 by mdevaev

akhamar, 2024-06-15:
Could you provide a bit more context and explanation on how one could use gpio menu for doing a physical clear CMOS jumper on a motherboard ?
That would be greatly appreciated.

Do you need extra hardware ? Can you just plug 2 wires from the clear CMOS to rpi gpio pin ?

realies, 2024-06-15:
I connect the jumper wires that usually go to the system reset header to the CMOS reset header.

akhamar, 2024-06-15:
> I connect the jumper wires that usually go to the system reset header to the CMOS reset header.

Usually clear CMOS need 5-10sec of long press, how do you achieve that with reset ?

realies, 2024-06-15:
> > I connect the jumper wires that usually go to the system reset header to the CMOS reset header.
> 
> Usually clear CMOS need 5-10sec of long press, how do you achieve that with reset ? As reset is just a short toggle.

The mobos I use it on don't require this.