pikvm/pikvm #1255: Auto switching KVM for Window:  how to run pikvm without video but force screen size?

jzhvymetal, 2024-02-19:
I found a new use case for PiKVM: controlling another computer and using it as an auto-switching KVM for Windows. This is achieved by adding a second virtual monitor and then using TigerVNC Viewer to force it into fullscreen mode on the virtual monitor. This method only works if you set the resolution the same as the video output on PiKVM. Is there any way with PiKVM to force a screen size without having any HDMI video?

Here are the steps for setting up an auto-switching KVM for Windows:

1.  Install the driver and secondary monitor (https://www.amyuni.com/forum/viewtopic.php?t=3030).
2.  Set up the monitor to match the resolution of the remote machine.
3.  Use TigerVNC to open fullscreen on the secondary monitor (Options -> Display -> Full screen on selected monitor).

jzhvymetal, 2024-02-19:
Guess I should of searched...will leave for others 

https://github.com/pikvm/pikvm/issues/351

jzhvymetal, 2024-02-19:
Okay,  the comment from https://github.com/pikvm/pikvm/issues/351 but it will work on the pikvm webpage but it will close tigervnc client as soon as it connects.  Also on the pikvm it is slow to open the fake display.  Any ideas?

jzhvymetal, 2024-02-20:
I found a work around not to force a close on tigervnc when there is no video stream **but there has to be a better way.**

1. Install kernel headers (I was using OrangePi so install headers on SBC being used) (Should compile)
```
apt install linux-headers-current-sunxi
```
2.  Install v4l2loopback kernel driver (Should compile)
```
apt install  v4l2loopback-dkms
```
3.  Install addional utils
```
apt install v4l2loopback-utils v4l-utils ffmpeg
```
4.  Install loopback device /dev/video5
```
modprobe v4l2loopback video_nr=5
``` 
5.  Edit /etc/kvmd/override.yaml and add some options:
```
kvmd:
    msd:
        type: disabled
    streamer:
        cmd_remove:
          - "--device=/dev/kvmd-video"
        cmd_append: 
          - "--device=/dev/video5"
          - "--blank=/etc/kvmd/blank.jpeg"
```
6.  Start ffmpeg stream
```
ffmpeg -f lavfi -i color=c=black@0.2:size=1920x1080 -vf "format=yuvj422p" -vcodec mjpeg -f v4l2 -loglevel error /dev/video5
``` 
7.  Restart kvmd
```
systemctl restart kvmd
```

mdevaev, 2024-03-16:
Hello. To be honest, I don't understand what's going on here at all. Is there any way you can show an example, or a video?

mdevaev, 2024-03-21:
Btw --blank option is deprecated now and will not be supported again because I've replaced it to custom text renderer. It looks like out of scope of the main use, it's better to connect to our [discord](https://discord.gg/bpmXfz5) to discuss this with other users.

closed 2024-03-21 by mdevaev