pikvm/pikvm #1258: Adding passkey support

panda2134, 2024-02-21:
**Is your feature request related to a problem? Please describe.**
Currently, if the user enables 2FA then a TOTP is required at every login. Given that some users do have access to devices that are more secure than an app-based 2FA code (e.g., security tokens like Google Titan or Yubikey), it would be great if PiKVM supports a passwordless login with those devices using a [passkey](https://passkey.org).

**Describe the solution you'd like**
Implement the passkey protocol and allow passwordless login with security tokens.

**Describe alternatives you've considered**
Do not implement the passkey protocol, but allow using WebAuthn / Yubikey as an alternative second factor. That is, instead of typing in the 6-digit code at every login, the user interacts with a security token to prove their identity.

**Additional context**
Try passkey at [https://passkey.org](https://passkey.org). I'm personally interested in submitting a pull request on this if I have time.

mdevaev, 2024-03-14:
Hello. I think it can be done, but most likely we will need a new plugin api for authorization inside KVMD.

upamanyudas, 2024-10-19:
this would be awesome to have! and also great that most password managers now (e.g. bitwarden) supports passkeys now, this would also streamline the login ux. making it faster.