pikvm/pikvm #1261: Keyboard+mouse doesn't work in UEFI on HP EliteBook

Sislik42, 2024-02-23:
Keyboard and mouse doesn't work in UEFI on HP EliteBook 840 14" G10 (Product number: 818Q7EA#UUZ).
When Windows boots, there is no issue, and keyboard and mouse is working properly.

We tried all USB A ports on mentioned NB as described in FAQ.
(We even tried USB on docking station, but we were expecting it will probably not work too.)

[image]

**Desktop (please complete the following information):**
 - OS: macOS
 - Browser: Safari
 - Version: 17.3.1 (19617.2.4.11.12)
 - VNC client: no

**PiKVM info:**
PiKVM V4 Plus, bought here https://www.pi-shop.ch/pikvm-v4-plus
PiKVM was just bought and updated to the latest version today
 - Raspberry Pi board version: ???
 - PiKVM platform: ???
 - Video capture type: ???
 - KVMD version:
kvmd 3.306-1
kvmd-fan 0.30-1
kvmd-oled 0.26-1
kvmd-platform-v4plus-hdmi-rpi4 3.306-1
kvmd-webterm 0.48-1
 - uStreamer version: ustreamer 5.51-1
 - Linux kernel: Linux pikvm 6.1.61-1-rpi-ARCH #1 SMP Fri Nov  3 20:48:52 MSK 2023 armv7l GNU/Linux

mdevaev, 2024-02-24:
Hello. For the first action, please try to disable mass storage drive: https://docs.pikvm.org/msd/#disabling-mass-storage

mdevaev, 2024-02-25:
:ok_hand:

jnstockley, 2024-03-02:
I am also having issues with my host device responding to input from my keyboard and/or mouse. I tried disabling the USB mass storage device, but that hasn't resolved the issue. Currently, TrueNAS Scale is installed, but also had an issue with the same device running Windows 10. When I unplug and re-plug the USB cable connected to the PiKVM I can see it's recognized. I am using a prebuilt PiKVM V3. I've also tried to re-install the OS which also didn't fix the issue.
[image]

mdevaev, 2024-03-03:
@jnstockley For the replugging case we have a special option. Switch it to OFF and back to ON. This is equivalent to reconnecting the cable, but can be done remotely in case of problems like yours.
[image]

jnstockley, 2024-03-08:
Thanks that is useful to know, but that doesn't seem to resolve the issue with the host not responding to keyboard and mouse interactions made through PiKVM. I'm assuming my issue is related to the original poster issue, but I could be wrong.

mdevaev, 2024-03-11:
@jnstockley What you see in dmesg when you're using usb switch in the menu?

mdevaev, 2024-03-11:
@Sislik42 You're welcome!

closed 2024-03-25 by mdevaev