pikvm/pikvm #1262: Raspberry Pi Zero 2 W Rev 1.0 <--- no wifi from brcmfmac43436*+

djieno, 2024-02-24:
**Describe the bug**
On a February 2024 purchased set of two raspberry pi zero 2 w  no wifi available with latest pikvm (nor debian after may 2023). Both hardware pi's work with an older version of debian [LINK](https://downloads.raspberrypi.org/raspios_oldstable_lite_armhf/images/raspios_oldstable_lite_armhf-2023-02-22/2023-02-21-raspios-buster-armhf-lite.img.xz) with my TPlink/AX3(Mikrotik)/WAP(Mikrotik).

**To Reproduce**
Steps to reproduce the behavior, like:
1. Use an pi zero 2 w with cat /proc/device-tree/model
2. verify with 
```
pi@picamhd:~ $ cat /proc/device-tree/model 
Raspberry Pi Zero 2 W Rev 1.0

pi@picamhd:~ $ dmesg | grep brcm
[    6.590884] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    6.612586] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436s-sdio for chip BCM43430/1
[    6.613346] usbcore: registered new interface driver brcmfmac
[    6.803657] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436s-sdio for chip BCM43430/1
[    6.803823] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436s-sdio for chip BCM43430/1
[    6.805886] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[    6.806647] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jul  5 2022 21:29:56 version 7.45.96.141 (g53ba0094) FWID 01-eef9b3bf es7
[    9.501377] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
```
Note **BCM43430/1** for version 1

**Expected behavior**
After flashing pikvm version using [LINK](https://docs.pikvm.org/flashing_os) and setting wifi credentials [LINK](https://docs.pikvm.org/on_boot_config/) it should connect to wifi and receive an ip from available dhcp server. 

**Screenshots**
No screen/weird hdmi conversion plug available

**Desktop (please complete the following information):**
 - OS: 22.04.3 desktop / using etcher or pi imager makes no difference

**PiKVM info:**
  - Raspberry Pi board version RPI zero 2 w 
  - PiKVM platform v2-hdmi]
  - Video capture type CSI bridge
  - Linux kernel: `uname -a` - can't access it but downloaded version is from February 2024 [LINK](https://files.pikvm.org/images/v2-hdmi-rpi4-latest.img.xz) 
**Additional context**
I noticed here [LINK](https://community.victronenergy.com/questions/144249/no-wifi-on-pi-zero-2w-venus-os-v29016-and-14.html) there might be issues with different version of the broadcom chips and the corresponding drivers for it. 'Never' had driver issues for mainstream hardware products with Linux since 2010. 

These file seem to be different located at /lib/firmware/brcm/
brcmfmac43436-sdio.txt brcmfmac43436s-sdio.bin brcmfmac43436s-sdio.txt

**Newer version that gives you grey hair**
awesomeuser@awesomedesktop:/media/awesomeuser/rootfs/lib/firmware/brcm$ ls -la brcmfmac43436*
-rw-r--r-- 1 root root 416101 nov 15 14:52 brcmfmac43436-sdio.bin
-rw-r--r-- 1 root root  11209 nov 15 14:52 brcmfmac43436-sdio.clm_blob
lrwxrwxrwx 1 root root     22 nov 15 14:52 brcmfmac43436-sdio.raspberrypi,model-zero-2-w.bin -> brcmfmac43436-sdio.bin
lrwxrwxrwx 1 root root     27 nov 15 14:52 brcmfmac43436-sdio.raspberrypi,model-zero-2-w.clm_blob -> brcmfmac43436-sdio.clm_blob
lrwxrwxrwx 1 root root     22 nov 15 14:52 brcmfmac43436-sdio.raspberrypi,model-zero-2-w.txt -> brcmfmac43436-sdio.txt
**-rw-r--r-- 1 root root   1706 nov 15 14:52 brcmfmac43436-sdio.txt
-rw-r--r-- 1 root root 442211 nov 15 14:52 brcmfmac43436s-sdio.bin**
lrwxrwxrwx 1 root root     23 nov 15 14:52 brcmfmac43436s-sdio.raspberrypi,model-zero-2-w.bin -> brcmfmac43436s-sdio.bin
lrwxrwxrwx 1 root root     23 nov 15 14:52 brcmfmac43436s-sdio.raspberrypi,model-zero-2-w.txt -> brcmfmac43436s-sdio.txt
**-rw-r--r-- 1 root root   1185 nov 15 14:52 brcmfmac43436s-sdio.txt**

**Older version that does work**
awesomeuser@awesomedesktop:/media/awesomeuser/rootfs/lib/firmware/brcm$ ls -la brcmfmac43436*
-rw-r--r-- 1 root root 416101 nov 15 14:52 brcmfmac43436-sdio.bin
-rw-r--r-- 1 root root  11209 nov 15 14:52 brcmfmac43436-sdio.clm_blob
lrwxrwxrwx 1 root root     22 nov 15 14:52 brcmfmac43436-sdio.raspberrypi,model-zero-2-w.bin -> brcmfmac43436-sdio.bin
lrwxrwxrwx 1 root root     27 nov 15 14:52 brcmfmac43436-sdio.raspberrypi,model-zero-2-w.clm_blob -> brcmfmac43436-sdio.clm_blob
lrwxrwxrwx 1 root root     22 nov 15 14:52 brcmfmac43436-sdio.raspberrypi,model-zero-2-w.txt -> brcmfmac43436-sdio.txt
**-rw-r--r-- 1 root root   1613 feb 24 14:24 brcmfmac43436-sdio.txt
-rw-r--r-- 1 root root 436733 feb 24 14:24 brcmfmac43436s-sdio.bin**
lrwxrwxrwx 1 root root     23 nov 15 14:52 brcmfmac43436s-sdio.raspberrypi,model-zero-2-w.bin -> brcmfmac43436s-sdio.bin
lrwxrwxrwx 1 root root     23 nov 15 14:52 brcmfmac43436s-sdio.raspberrypi,model-zero-2-w.txt -> brcmfmac43436s-sdio.txt
**-rw-r--r-- 1 root root   1092 feb 24 14:24 brcmfmac43436s-sdio.txt**

After digesting reality and spending quite some time on this 'easy task to use an pikvm' I realized my hardware version of pi zero 2 w would be future doomed anyway. There decided to return the pi's and contact Raspberry makers with my annoyances.

Also checked out these threads 
didn't resolve the issue - not working for me

over_voltage=2 didn't work
different key_mgt
different cables didn't make it work
wifi country to GB/US
https://github.com/pikvm/pikvm/issues/137
https://github.com/pikvm/pikvm/issues/181
https://github.com/pikvm/pikvm/issues/194 
https://github.com/pikvm/pikvm/issues/1028
github://https.pikvm/com/pikvm/issues/911

**Would be solution**
Have backward compatible drivers etc and symlink appropriately by establishing hardware version. This would make hardware future proof and saves us from piling 'redundant' hardware.

mdevaev, 2024-02-26:
Sup. I'll update kernel and we'll check if it's working.

djieno, 2024-02-26:
> Sup. I'll update kernel and we'll check if it's working.

If you need met to test something - I'm happy to help out

mdevaev, 2024-03-14:
Sup. Sorry for delay. You can check the fresh kernel 6.6, it's already in the repo.

mdevaev, 2024-03-14:
I've also released new v2-hdmi-zero2w image, so you can flash it and check.

mdevaev, 2024-03-16:
Ok, we've checked the latest update and it's working fine with same zero2 device.

closed 2024-03-16 by mdevaev

rafaelreis-r, 2024-05-16:
This is still going on the latest images. I had to roll back to an image from March to get wifi up again on 2 zero2w I have here deployed. 

As soon as I do pikvm-update and reboot, it loses connection, and display shows <no-ip> and <no-iface>

Since there is no console, or lan connectivity, how can I debug this besides reflashing?

jangrewe, 2024-07-25:
@mdevaev I can confirm that this is still an issue after running `pikvm-update`. I flashed an SD for a Pi Zero 2W with the latest image, it boots fine and everything works, but after updating the Wifi is gone.

I figured it out via the USB-to-Serial console, thanks for that...

mdevaev, 2024-07-25:
Did you try fresh image of the official Raspberry OS with latest updates? We need to understand this is an Arch or Raspberry upstream problem.

mdevaev, 2024-07-27:
Please follow here: #1375