pikvm/pikvm #1264: Building PiKVM OS not pissible due to crosscompiling error

wsxws, 2024-02-25:
https://docs.pikvm.org/building_os/ explains how to build an os and example uses apt-get like working on debian BUT
after installing all dependencies and creating config.mk and executing make os it leads to:

Makefile:64: *** Cross-arch ARM building like aarch64<->arm is not supported.  Stop.

but an PI 4 even using a 32bit RaspiOS (bullseye or bookworm doesnt matter) is on aarch64, how to solve that ?

mdevaev, 2024-02-25:
Hello. Probably you have 64 bit OS. Run `arch` and `uname -a` commands on it.

wsxws, 2024-02-25:
if i'd see such a silly question i would think the same.
i already checked it 2 times, but i just did it a thir time.
[image]
and when i log in it says:
[image]
any other idea ?

mdevaev, 2024-02-25:
Well, it clearly states here that your OS is aarch64. Maybe there is something wrong with the links in the Imager. Download the OS directly from the Raspberry website. If the Imager is really pumping 64 bits to a 32-bit request, you should let them know about this bug.

wsxws, 2024-02-25:
hm
some funny facts
i also tried out tinypilot. and as curious as i am i first tried to install it on 64bit, what did not work out. so i installed the 32bit version and installation of tinypilot ran throught and is working. when i log in on this system uname also shows arch64. (whysoever)

anyway i meanwhile managed to install pikvm v3 with hat on pi 4 using raspi os bookworm 64bit and also installed v2 with usb dongle on pi5 using bookworm 64bit as well. just display-services are missing atm, ill add them lateron

thanks for your great project over here

closed 2024-02-25 by wsxws