pikvm/pikvm #1266: Support for HDPI Displays

makuru-dd, 2024-03-03:
**Is your feature request related to a problem? Please describe.**
I Have a new laptop and that has a 2k display and it is almost impossible to read the text in the terminal of the web interface.

**Describe the solution you'd like**
A button or scroll wheel to designate a text size for the terminal to use.

**Describe alternatives you've considered**
I tried to zoom in with the Browser but that did very little but make all the borders bigger.

[image]

mdevaev, 2024-03-14:
Hello. Which browser do you use?

mdevaev, 2024-03-20:
Sup?

makuru-dd, 2024-03-20:
Hi I tried it with Librewolf (firefox fork based on web privacy, with a focus against fingerprinting) and ungoogled chromium.

makuru-dd, 2024-03-20:
Seems like a update fixed it, I updated the PIKVM to the second last version and it seems to have fixed it.

mdevaev, 2024-03-20:
Okay, then I'm closing it :ok_hand:

closed 2024-03-20 by mdevaev