pikvm/pikvm #1270: [Feature Request] pikvm-update option to prevent automatic reboot

n1nj4888, 2024-03-09:
**Is your feature request related to a problem? Please describe.**
The current `pikvm-update` command does not have any options to control the behavior and so, if a reboot is determined as required, the command performs the reboot automatically (albeit asking the user to `ctrl+c` within 30 seconds if they want to abort the reboot).

For scenarios where we might want to call the pikvm-update command via say an Ansible playbook, the automatic reboot causes the playbook to fail as it is not expected. As such, it would be great to have a `--noreboot `option so that, even if a reboot is determined, pikvm-update would still output to stdout that a reboot is required, but would not actually perform the reboot itself, leaving the decision of when to reboot to the user or another orchestrating process (such as Ansible)

Thanks for your consideration!

**Describe the solution you'd like**
Please add an option for `--noreboot` to the pikvm-update command.

**Describe alternatives you've considered**
N/A

**Additional context**
Add any other context or screenshots about the feature request here.

mdevaev, 2024-03-10:
I've added option `--no-reboot`, but should we use some special exit code like `111` which can means `reboot required` for caller?

n1nj4888, 2024-03-10:
Hi @mdevaev ! Thanks for looking into this!

I suspect the caller could just grep the command stdout for some specific indicator (i.e "Reboot required but not performed" or similar) but a special exit code would be better (although I'm not familiar with the conventions here) and make it easier to determine the follow-on action.

Thanks!

mdevaev, 2024-03-10:
Ok, exit code == 100 means "reboot required" now if option `--no-reboot` specified. Released.

closed 2024-03-10 by mdevaev