pikvm/pikvm #1272: Google Authenticator for SSH is broken after the last update

DigitalYemeni, 2024-03-09:
**Describe the bug**
My PiKVM was working fine with the Google Authenticator for 2FA for SSH sessions. After updating the PiKVM to the last version, i could not access the ssh from outside the Web as before. It always gives me authentication failure error until I disable the google-authenticator settings in /etc/pam.d/sshd and /etc/ssh/sshd_config.d/99-archlinux.conf 

**To Reproduce**
Steps to reproduce the behavior, like:
install libpam-google-authenticator and configure it accordingly.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari] all
 - Version [e.g. 22]
 - VNC client (if used)

**PiKVM info:**
kvmd 3.310-1
kvmd-fan 0.30-1
kvmd-oled 0.26-1
kvmd-platform-v2-hdmi-rpi4 3.310-1
kvmd-webterm 0.48-1

  - Raspberry Pi board version RPi 4
  - PiKVM platform [e.g. v2-hdmi]
  - Video capture type: CSI bridge
  - KVMD version: `pacman -Q | grep kvmd`
  - uStreamer version: ustreamer 6.0-1
  - Linux kernel: `uname -a`

**Additional context**
Add any other context about the problem here.

mdevaev, 2024-03-09:
Hello. Which manual *exactly* you used to setup it? I need it because the steps can be different.

DigitalYemeni, 2024-03-09:
I used this one described here:

https://wiki.archlinux.org/title/Google_Authenticator

Used to work well but now broken!

Thank you!!

mdevaev, 2024-03-09:
Can you reach to the shell via web terminal?

DigitalYemeni, 2024-03-09:
yes, I can still use the terminal from Web, in fact, when i enabled the google 2FA, it does not ask me for the OTP. Which works in my favor as i already configure the OTP for the Web. 
It used to ask me for the ssh from external (i.e., windows ssh or putty, etc) 
The Web was not affected by installing the Google 2FA, and that's how I disabled the google it to find the root cause of the ssh issue.

mdevaev, 2024-03-09:
Please attach the log: `journalctl -u sshd`

DigitalYemeni, 2024-03-09:
here it is:

[root@pikvm kvmd-webterm]# journalctl -u sshd
Mar 05 01:43:08 pikvm systemd[1]: Started OpenSSH Daemon.
Mar 05 01:43:08 pikvm sshd[427]: Server listening on 0.0.0.0 port 22.
Mar 05 01:43:08 pikvm sshd[427]: Server listening on :: port 22.
Mar 09 09:56:59 pikvm sshd(pam_google_authenticator)[1051]: Invalid verification code for root
Mar 09 09:57:01 pikvm sshd[1051]: Failed password for root from 172.x.x.x port 63579 ssh2
Mar 09 09:57:07 pikvm sshd[1051]: error: Received disconnect from 172.x.x.x port 63579:13: The user canceled authentication.  [preauth]
Mar 09 09:57:07 pikvm sshd[1051]: Disconnected from authenticating user root 172.x.x.x port 63579 [preauth]
Mar 09 10:09:08 pikvm sshd[427]: Received signal 15; terminating.
Mar 09 10:09:08 pikvm systemd[1]: Stopping OpenSSH Daemon...
Mar 09 10:09:08 pikvm systemd[1]: sshd.service: Deactivated successfully.
Mar 09 10:09:08 pikvm systemd[1]: Stopped OpenSSH Daemon.
Mar 09 10:09:09 pikvm systemd[1]: Started OpenSSH Daemon.
Mar 09 10:09:09 pikvm sshd[2040]: Server listening on 0.0.0.0 port 22.
Mar 09 10:09:09 pikvm sshd[2040]: Server listening on :: port 22.
Mar 09 10:09:19 pikvm sshd(pam_google_authenticator)[2054]: Invalid verification code for root
Mar 09 10:09:19 pikvm sshd(pam_google_authenticator)[2054]: Invalid verification code for root
Mar 09 10:09:21 pikvm sshd[2054]: Failed password for root from 172.x.x.x port 63646 ssh2
Mar 09 10:09:30 pikvm sshd(pam_google_authenticator)[2054]: Invalid verification code for root
Mar 09 10:09:30 pikvm sshd(pam_google_authenticator)[2054]: Invalid verification code for root
Mar 09 10:09:33 pikvm sshd[2054]: Failed password for root from 172.x.x.x port 63646 ssh2
Mar 09 10:11:13 pikvm sshd[2054]: fatal: Timeout before authentication for 172.x.x.x port 63646
[root@pikvm kvmd-webterm]#

DigitalYemeni, 2024-03-09:
Here is my installation log and how i configured it if you want to reproduce it:

+++++++++++++++++++++
Full installation Log
+++++++++++++++++++++

[root@pikvm ~]# pacman -S libpam-google-authenticator
resolving dependencies...
looking for conflicting packages...

Packages (1) libpam-google-authenticator-1.09-1

Total Download Size:   0.03 MiB
Total Installed Size:  0.08 MiB

:: Proceed with installation? [Y/n] y
:: Retrieving packages...
 libpam-google-authenticator-1.09-1-armv7h
(1/1) checking keys in keyring                                            [#########################################] 100%
(1/1) checking package integrity                                          [#########################################] 100%
(1/1) loading package files                                               [#########################################] 100%
(1/1) checking for file conflicts                                         [#########################################] 100%
(1/1) checking available disk space                                       [#########################################] 100%
:: Processing package changes...
(1/1) installing libpam-google-authenticator                              [#########################################] 100%
Optional dependencies for libpam-google-authenticator
    qrencode: scannable QR codes for google auth phone app
:: Running post-transaction hooks...
(1/1) Arming ConditionNeedsUpdate...
[root@pikvm ~]# google-authenticator

Do you want authentication tokens to be time-based (y/n) y
Warning: pasting the following URL into your browser exposes the OTP secret to Google:
  https://www.google.com/chart?chs=200x200&chld=M|0&cht=qr&chl=otpauth://totp/root@pikvm%3Fsecret%3DWVMKANNL7PRW42UPXxxxxxxxxxxx%26issuer%3Dpikvm
Failed to use libqrencode to show QR code visually for scanning.
Consider typing the OTP secret into your app manually.
Your new secret key is: WVMKANNL7PRW42UPxxxxxxxxx
Enter code from app (-1 to skip): 9167xx
Code confirmed
Your emergency scratch codes are:
xxx will not work anyways due to ReadOnly mode!

Do you want me to update your "/root/.google_authenticator" file? (y/n) y

Do you want to disallow multiple uses of the same authentication
token? This restricts you to one login about every 30s, but it increases
your chances to notice or even prevent man-in-the-middle attacks (y/n) n

By default, a new token is generated every 30 seconds by the mobile app.
In order to compensate for possible time-skew between the client and the server,
we allow an extra token before and after the current time. This allows for a
time skew of up to 30 seconds between authentication server and client. If you
experience problems with poor time synchronization, you can increase the window
from its default size of 3 permitted codes (one previous code, the current
code, the next code) to 17 permitted codes (the 8 previous codes, the current
code, and the 8 next codes). This will permit for a time skew of up to 4 minutes
between client and server.
Do you want to do so? (y/n) y     

If the computer that you are logging into isn't hardened against brute-force
login attempts, you can enable rate-limiting for the authentication module.
By default, this limits attackers to no more than 3 login attempts every 30s.
Do you want to enable rate-limiting? (y/n) n
[root@pikvm ~]# 
[root@pikvm ~]# nano /etc/pam.d/sshd
add the last line:

 #%PAM-1.0
 2
 3 auth      include   system-remote-login
 4 account   include   system-remote-login
 5 password  include   system-remote-login
 6 session   include   system-remote-login
 7 auth      required   pam_google_authenticator.so

cat /etc/ssh/sshd_config.d/99-archlinux.conf

  GNU nano 7.2
# sshd_config defaults on Arch Linux
KbdInteractiveAuthentication yes
UsePAM yes
PrintMotd no

Note: I removed the commended lines in the /etc/ssh/sshd_config:

[root@pikvm ~]# cat /etc/ssh/sshd_config

ChallengeResponseAuthentication yes
AuthorizedKeysFile      .ssh/authorized_keys
KbdInteractiveAuthentication no
UsePAM yes
Subsystem       sftp    /usr/lib/ssh/sftp-server
PermitRootLogin yes

[root@pikvm ~]# 
[root@pikvm ~]# systemctl restart sshd
[root@pikvm ~]# reboot
Broadcast message from root@pikvm on pts/2 (Sat 2023-10-21 15:05:59 MSK):

The system will reboot now!

mdevaev, 2024-03-09:
Please verify the system time if it is correct: `timedatectl`

DigitalYemeni, 2024-03-09:
[root@pikvm(g:main) g:/]# timedatectl
               Local time: Sat 2024-03-09 11:02:32 UTC
           Universal time: Sat 2024-03-09 11:02:32 UTC
                 RTC time: n/a
                Time zone: UTC (UTC, +0000)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no
          
it used to work with the same time zone settings. do you recommend to re-configure it?

mdevaev, 2024-03-09:
Apparently, this error is not on our part, but somewhere in the upstream distribution. Please contact the Arch developers with this. https://archlinuxarm.org/forum/

DigitalYemeni, 2024-03-09:
Noted. Many thanks!

mdevaev, 2024-03-09:
I'm sorry I can't help you more.

closed 2024-03-09 by mdevaev

senyahnoj, 2024-05-08:
I had the same problem on a RHEL system. The solution was to install an additional package called **qrencode** which contains the missing libs