pikvm/pikvm #1273: ATX Short Press too short; ATX Long Press Too long

FeintDoxx, 2024-03-11:
Perhaps there is a way around this and I am not aware of it, but as title states.

I have the GPIO board connected to a power button on a dock, only was able to figure out the button.

When i send the ATX short press the ui shows it is sending the input but no response from the dock, if I send a long press then I can see the light on the docking station come up indicating power on attempt but then it stays off because it's still being pressed and doesn't turn on.

Tested the button itself, still woks but seems like the length that needs to be held to power on is the length of 2 short presses, tried using:

`curl -X POST -k -u admin:admin https://<pikvm-ip>/api/atx/click?button=power`

but then I get 
`{
    "ok": false,
    "result": {
        "error": "ForbiddenError",
        "error_msg": "Forbidden"
    }
}`

behind an apache server and 2fa setup:

`<VirtualHost *:80>
2           ServerName pikvm.local
  1         Redirect permanent / https://192.168.1.2/
  2 </VirtualHost>
  3
  4 <VirtualHost *:443>
  5         ServerAdmin admin@test.com
  6         ServerName https://pikvm.local/
  7
  8         Redirect permanent / https://192.168.1.2/
  9
 10         SSLProxyEngine On
 11         SSLProxyVerify none
 12         SSLProxyCheckPeerCN off
 13         SSLProxyCheckPeerExpire off
 14
 15
 16         ProxyPass / https://192.168.1.2/
 17         ProxyPassReverse / https://192.168.1.2/
 18
 19         <Location /api/ws>
 20                 Order allow,deny
 21                 Allow from all
 22                 ProxyPass wss://192.168.1.2/api/ws
 23                 ProxyPassReverse wss://192.168.1.2/api/ws
 24         </Location>
 25
 26         <Location /janus/ws>
 27                 Order allow,deny
 28                 Allow from all
 29                 ProxyPass wss://192.168.1.2/janus/ws
 30                 ProxyPassReverse wss://192.168.1.2/janus/ws
 31         </Location>
 32
 33         SSLEngine on
 34         SSLCertificateFile /etc/ssl/test.crt
 35         SSLCertificateKeyFile /etc/ssl/private/test.key
 36         SSLCertificateChainFile /etc/ssl/test.crt
 37
 38         ErrorLog ${APACHE_LOG_DIR}/fgkvm-error.log
 39         CustomLog ${APACHE_LOG_DIR}/fgkvm-access.log combined
 40 </VirtualHost>
`

Am I just missing something in apache to allow the POST attempts to go through? Or is there another way to change the short press length? Thanks either way.

mdevaev, 2024-03-12:
I don't know about Apache, but you can adjust the duration of the click in `/etc/kvmd/override.yaml` (in seconds):
```yaml
kvmd:
    atx:
        click_delay: 0.1
        long_click_delay: 5.5
```
After changing, restart the service: `systemctl restart kvmd`

FeintDoxx, 2024-03-12:
That most definitely did the trick; Very sorry, just realized this was listed in the `kvmd -m` command. Thank you, was a long day.

closed 2024-03-12 by FeintDoxx

mdevaev, 2024-03-12:
:ok_hand: