pikvm/pikvm #1275: No UEFI/boot output on the Intel NUC 9 Pro Kit NUC9VXQNX with mini PiKVM v4

medecigo3, 2024-03-13:
Hi developers, 

I'm unable to see video whilst the mini PiKVM v4 is connected to the HDMI port of my Intel NUC 9 Pro Kit NUC9VXQNX.

Can you please help debug this problem? Thanks in advance. 

Regards, 
Arturo.

mdevaev, 2024-03-14:
Hello! Is this a latest version of PiKVM OS? If not, please update it: `curl https://files.pikvm.org/update-os.sh | bash`.

After that, use command `rw; kvmd-edidconf --restore-default=v4mini; reboot`

It should help because we fixed display EDID.

medecigo3, 2024-03-14:
Hi @mdevaev , thanks for your help. 

Indeed I was missing the OS update. I can confirm my mini PiKVM v4 has working video, mouse and keyboard with my Intel NUC 9 Pro Kit NUC9VXQNX.

Regards, 
Arturo.

mdevaev, 2024-03-14:
Glad to hear it. You're welcome!

mdevaev, 2024-03-14:
Did you try `System -> Reset stream`?

medecigo3, 2024-03-14:
@mdevaev , Thanks for the tip. 

I've now tried the `System -> Reset stream`  several times however the issue still persists, I see a screen with static noise. Screenshot attached. 
[image]

Regards, 
Arturo.

mdevaev, 2024-03-14:
Could you take a photo of your setup?

medecigo3, 2024-03-14:
Please see my setup in the screenshots attached:
[image]
[image]
[image]
[image]
[image]

mdevaev, 2024-03-14:
Thank you. Looks fine for me. There is another recipe to fix the connection. Please check it:
1. Close the web page.
2. Wait ~15 seconds
3. `systemctl stop kvmd-tc358743`
4. Wait ~5 seconds
5. `systemctl start kvmd-tc358743`
6. Open the stream.

medecigo3, 2024-03-14:
Unforntunately the last recipe fix did't solve the screen with static noise. 
Regards, 
Arturo.

mdevaev, 2024-03-15:
Got it. Please wait a bit, we'll debug it on our side today.

mdevaev, 2024-03-15:
What happens if you make `systemctl stop kvmd-tc358743` and after that open the stream?

medecigo3, 2024-03-15:
Hi @mdevaev , 

I've tried with `systemctl stop kvmd-tc358743`. Afterwards, I tried close/open the web interface and/or `System -> Reset stream` but a the message < NO SIGNAL > appears on the screen. What are the steps to open the stream? 
Regards, 
Arturo.

mdevaev, 2024-03-15:
After stopping, when you have `NO SIGNAL`, please run `v4l2-ctl --log-status` and save this log. Run `systemctl start kvmd-cc358743` while you have the `NO SIGNAL` stream, wait for the picture and run `v4l2-ctl --log-status` again. Describe what you see and attach both logs.

medecigo3, 2024-03-15:
Please find the logs attaached after running `v4l2-ctl --log-status` while `no signal` message is  disaplyed on KVM web interface. 
[v4l2-ctl --log-status.log](https://github.com/pikvm/pikvm/files/14620549/v4l2-ctl.--log-status.log)

The `systemctl start kvmd-cc358743` command generated the following error  while `no signal` is displayed on the KVM web interface: 
`Failed to start kvmd-cc358743.service: Unit kvmd-cc358743.service not found.`

mdevaev, 2024-03-15:
Typo. `systemctl start kvmd-tc358743`

medecigo3, 2024-03-16:
@mdevaev , plese see the debugging steps below. 

1. Video is lost:
[image]
2.  Close the web page.
4. Wait ~15 seconds
5. `systemctl stop kvmd-tc358743` See logs.txt
6. Wait ~5 seconds
7. Open Web page and start KVM session
8. NO SIGNAL 
9. `systemctl start kvmd-tc358743` See logs.txt
10. Result:
[image]

Logs: 
[logs.txt](https://github.com/pikvm/pikvm/files/14623317/logs.txt)

mdevaev, 2024-03-16:
> Logs:
> [logs.txt](https://github.com/pikvm/pikvm/files/14623317/logs.txt)

Is this a log after all steps, when static noise is back?

mdevaev, 2024-03-16:
Ah, no, I see, Please make log after the starting tc358743 too.

medecigo3, 2024-03-19:
@mdevaev m please find the logs attached: 
[logs.txt](https://github.com/pikvm/pikvm/files/14646112/logs.txt)

Steps:
1. Video is lost:
[image]
2. Close the web page.
3. Wait ~15 seconds
4. systemctl stop kvmd-tc358743
5. Wait ~5 seconds
6. Open Web page and start KVM session
7. NO SIGNAL
[image]
8. v4l2-ctl --log-status
9. systemctl start kvmd-tc358743
10. v4l2-ctl --log-status

mdevaev, 2024-03-19:
And after `systemctl start kvmd-tc358743` you see the noise again, right?

medecigo3, 2024-03-19:
Correct, after `systemctl start kvmd-tc358743`  is executed, the static noise still appears in the video stream.

mdevaev, 2024-03-19:
Got it. We'll check possible buggy places with this scenario.

medecigo3, 2024-03-22:
Please find logs with a new test attached:

[log.txt](https://github.com/pikvm/pikvm/files/14722034/log.txt)

Steps to reproduce the issue:

1. Video is lost. 
2. Close the web page.
3. Wait ~15 seconds
4. `systemctl stop kvmd-tc358743`
5. Wait ~5 seconds
6. Open Web page and start KVM session
7. NO SIGNAL screen appears
8. `v4l2-ctl --log-status`
9. `systemctl start kvmd-tc358743`
10. `v4l2-ctl --log-status`
11. Wait ~5 mins.
12. Video stream not restored:
[image]

mdevaev, 2024-03-22:
Can you try different HDMI cable?

medecigo3, 2024-03-22:
Hi @mdevaev , what are the HDMI cable specs that you would recommend ?

mdevaev, 2024-03-22:
Just try different cables that you have at home. There is no need to buy anything, at least for the test.

medecigo3, 2024-04-09:
Hi @mdevaev , I've tried with different cables however I'm facing the same issue. The quickest solution seems to be to disconnect and re-connect the HDMI cable. 

Regards, 
Arturo.

medecigo3, 2024-04-19:
Hello @mdevaev , can you please confirm what's the expected behavour with other devices? After a moment of inactivity, should the PiKVM automatically establish the video or should this trigger a manual action to re-establish the video stream?
Regards, 
Arturo.

mdevaev, 2024-04-19:
This is not expected behavour. And it's not reproducible on our test hosts. We'll checking soft/hardware now.

medecigo3, 2024-05-05:
@mdevaev , can we compare our logs when video stream is re-established after a period of inactivity? In case the HDMI stream isn't restablished correctly, should we expect to see errors in the logs? Is there a communicaiton protocol with error handling mechanism for streaming video content? 
Regards, 
Arturo.

mdevaev, 2024-05-07:
It seems to be a hardware incompatibility with your particular NUC model :/ I haven't figured out how to reproduce this on my side yet.

medecigo3, 2024-05-12:
Hi @mdevaev , it would be difficult to reproduce the issue on your side. What would be the best practices in this situation? 

- HDMI cable specifications? 
- Close the web page imediately when not in use? 
- script to restart the PIKVM? 

My NUC also features graphics card NVIDIA Quadro P2200 special NUC9 version which supports 4 x DisplayPorts. the NUC ins't running with this card. How well would the PIKVM support video if using a DP -> HDMI adapter connected to the PIKVM ?  

Regards, 
Arturo.

mdevaev, 2024-05-14:
I only have a couple of reports about a similar problem, so I don't really understand what's going on here. Judging by the logs and indirect debugging results, the problem is that NUC does not strictly follow the HDMI specifications in terms of connection management. Right now I can't do anything about it, just advise not to turn off the screen and not use various energy-saving modes. Sorry.