pikvm/pikvm #1276: h264 / webrtc not working on ungoogled-chromium

realies, 2024-03-16:
this has been going on for probably more than a year, but using ungoogled-chromium on either windows or macos can't connect via h264 / webrtc; it appears that it handshakes with janus but the screen is black and bitrate at 0kbps; any ideas what could be going wrong?

[image]

mdevaev, 2024-03-16:
Hello. Try to use alternative STUN server: https://docs.pikvm.org/webrtc/#how-its-working

mdevaev, 2024-03-16:
We have tested ungoogled chromium with static WebRTC configuration in the local network without any STUN server, and it's not working too. It looks like the guys from ungoogle team broke off something necessary besides unnecessary.

The problem is not on our side, so please redirect this issue to the ungoogled chromium team. I'm sorry, but there's nothing I can do here.

closed 2024-03-16 by mdevaev

Snuupy, 2024-03-17:
since we're talking about broken webrtc I've noticed it's also broken in librewolf for a long time, haven't on latest but I'll do so and report back

mdevaev, 2024-03-17:
Okay. The problem may be identical. We don't use any specific things that Google would require. I use regular Chromium and it works great. Other forks sometimes break something, for example, Vivaldi did not work for my colleague before.

Snuupy, 2024-03-29:
I tested webrtc on librewolf and it's working. https://mozilla.github.io/webrtc-landing/gum_test.html

When I switch to webrtc on librewolf for pikvm, I get this error:

```
Ooops: 465 Error processing SDP [janus.js:1434:13](https://snupikvm.lan/share/js/kvm/janus.js)
    transaction https://snupikvm.lan/share/js/kvm/janus.js:1434
    handleEvent https://snupikvm.lan/share/js/kvm/janus.js:770
```

I also tried following these instructions https://jrballesteros05.codeberg.page/posts/librewolf/ to enable webrtc. Is there any other debugging you want me to do?

It is working on firefox.

mdevaev, 2024-03-29:
465 means that the browser has transmitted an incorrect SDP. This is really a problem inside the browser. Librewolf devs should compare the SDP that their browser generates with what Chrome generates. It is likely that some kind of discrepancy will be found leading to problems.

realies, 2024-06-18:
This issue gets fixed on Ungoogled Chromium by going to `chrome://flags/#webrtc-ip-handling-policy` and changing `WebRTC IP policy` from `Disable non proxied udp` to `Default`.

Snuupy, 2024-06-18:
not that this is pikvm's fault, but if anyone is interested - on a related note, webrtc is likely not working in librewolf because https://mozilla.github.io/webrtc-landing/pc_test_no_h264.html shows no h264 support (on chrome/firefox it does)